LukeSkywalker92/TeleFrame #2: Switch off screen at night

LukeSkywalker92, 2018-09-06:


parachutesj, 2019-10-23:
Does this feature work? I tried it with the stock 7" touch display and it does not switch off

LukeSkywalker92, 2019-10-23:
At the moment this feature is only working for screens connected via hdmi.
I'm not sure if there is a command to switch off the stock 7" touch display.

simonreinhardt, 2019-10-25:
this shoud work with the stock 7" touch display: 
xset -display :0 dpms force off
xset -display :0 dpms force on

LukeSkywalker92, 2019-10-25:
Maybe one of you can verify if this is working by changing the commands in lines [29](https://github.com/LukeSkywalker92/TeleFrame/blob/562e851be7326a7840c68af69059c4eaf7036701/js/schedules.js#L29) and [37](https://github.com/LukeSkywalker92/TeleFrame/blob/562e851be7326a7840c68af69059c4eaf7036701/js/schedules.js#L37) of schedules.js.

Worren, 2019-11-17:
Yes this works. But the background light is still on.
Better is:
sudo sh -c 'echo 0 >> /sys/class/backlight/rpi_backlight/bl_power'
and for off:
sudo sh -c 'echo 1 >> /sys/class/backlight/rpi_backlight/bl_power'

nautical-miles, 2019-11-18:
I tried the software with a 15.6" touchscreen attached to a raspberry pi.
Turning off worked like a charm. Turning on made the screen blink for an hour until the turn on hour was over.
Instead of "tvservice --preferred & sudo chvt 6 && sudo chvt7" I prefer to use: 
"vcgencmd display_power 1"
for turning on and
"vcgencmd display_power 0" for turning off.
After a few seconds also the background light goes off.

LukeSkywalker92, 2019-11-20:
Ok, so there are many different commands depending on your specific setup. What do you think about config parameters, which allow to set the command for turning on and off the screen? This sure is not very comfortable for beginners, but we could add some examples to the README.

mistau, 2019-11-20:
I actually thought about putting the command to switch the display on/off into the config itself but decided against that one as it makes the config much more complicated and prone to errors.
What makes the approach to specify the switch command in the config even more ugly is the fact that the permissions on the device node controlling the backlight has to be adjusted after each boot. So you either need another config value holding the command to initialize the backlight control or change the permissions with the switch commands itself which makes them even worse.
For completeness: the other idea I had was to start shell scripts controlling the display - I gave up on that as I did not see an advantage to substitute one script by another, just another indirection. (we could let the config variable point to a script which accept the argument "on" or "off")
The gating factor for me to split into HDMI / other was the understanding that we basically have to cope with HDMI or DSI displays which is covered by a simple bool variable.
In case there are DSI displays which are switched of differently I would introduce a DsiDisplayVariant variable, as of now I am not aware of any.
Do you have more insight what other displays are around?

sohamakl, 2019-11-23:
I'm using a very nice 10.1'' display from sunfounder [sunfounder](http://wiki.sunfounder.cc/index.php?title=10.1_Inch_Touch_Screen_for_Raspberry_Pi).

Switching off the HDMI signal works without any problems. Unfortunately the display shows "No Signal" permanently. Does anyone have an idea how to switch the display to standby mode via command line?

LukeSkywalker92, 2019-11-23:
I don’t think this is possible. You cannot send commands like this via HDMI. Usually you can configure such things in the settings of HDMI screens.

sohamakl, 2019-11-23:
I found a workaround after I came across the following [link.](https://pi-buch.info/hdmi-display-aus-und-wieder-einschalten/)

In principle I do the following:

#### switch off ####
tvservice -o # switch off hdmi, no signal will apear after a while
tvservice -p # sent this directly after tvservice -o to switch the monitor on again. The display will be black until you switch to any virtual console and back to usually 7 again

#### switch on ####
chvt 1 # switch to virtual console 1
chvt 7 # and back to virtual console 7

To keep it simple I followed the example and controll the screen by cron job. Enclosed my adaptation of the [hdmi](https://github.com/LukeSkywalker92/TeleFrame/files/3882931/hdmi.txt) init.d script.

mistau, 2019-12-01:
as my pull request got more or less stuck in the system I thought of a new variant how to get this into the general source tree. I implemented my new approach already but would like to receive feedback before I start testing this and open another pull request:
General idea is to introduce config files for the screens containing all relevant settings describing a screen (pixels x/y, can switch off backlight, can dim backlight, has touch....) as well as the commands to perform actions like switching on/off the backlight. (dimming i something which I would like to see as well - I would therefore enable this so it could be added to the new touch menu).
In the general config (config/config.js) there is just a pointer to the according screen config file (I kept them all in a separate directory). On top there is a default config in case the according entry is missing in the config/config.js pointing to the HDMI screen/commands which are currently used.

For me this is actually a very flexible but still easy to configure approach, over time we might get the configs for the most common screen models. On top it doesn't even break the old config files.
Please share your thoughts.

sohamakl, 2019-12-01:
Sounds got for me.

nautical-miles, 2019-12-05:
For my combination of computer and screen (raspberry 3 and 7" touch display) the commands in schedule.js are not working. I set the schedule yesterday to 8:00 for start and 17:00 for end. When I came back this morning before 8 the screen was still running. The vcgencmd command mentioned before is working fine for my setup. 
Checking the logs I found a little glitch. The turn on and off commands are repeated for a full hour in the schedule. Sending the command for 1 second should be enough. Replacing the first two asterisks in in the schedule job by numbers for second and minute did the trick.
Can someone please check this and maybe change it in the master?
Thanks

mistau, 2019-12-05:
There is a pull request I contributed already in github which works for the official Raspberry Pi 7' screen connected via DSI. It adds another variable in the config which was considered not optimal, I am now working on a more general approach using dedicated config files for the screen settings, I guess this will make it earliest to a new major version. See above thread for details.

In case you are using the Raspberry DSI 7' display and need a quick solution apply the patch I provided in the pull request https://github.com/LukeSkywalker92/TeleFrame/pull/32 and add the additional config value to config/config.js (hdmiScreen: false,). This should fix your problem.

Trenar, 2019-12-05:
> I'm using a very nice 10.1'' display from sunfounder [sunfounder](http://wiki.sunfounder.cc/index.php?title=10.1_Inch_Touch_Screen_for_Raspberry_Pi).

Hi @sohamakl . I have the very same display and encounter some really strange issues with the wifi.. If I connect my RPi 3A+ to the power output of the display, I can not set up stable wifi connections. Having a dedicated power supply it works seamlessly. Do you have similar experiences and / or solutions?

nautical-miles, 2019-12-05:
Sorry, I can't help you with that. The 7" display I am using is not the official Raspberry Pi 7" display connected via DSI. It is a cheap one connected via hdmi.

mistau, 2019-12-05:
@nautical-miles forget my comment with the pointer to the patch, this only works for the official Raspberry DSI display.
Still the fact that the current HDMI-style way to switch off the display doesn't work for you is another argument to switch to configuration files for specific displays. Stay tuned.

sohamakl, 2019-12-05:
> Hi @sohamakl . I have the very same display and encounter some really strange issues with the wifi.. If I connect my RPi 3A+ to the power output of the display, I can not set up stable wifi [...]

I have no issue with wifi but I’m going to test a separate power supply to see if freeze issue will change.

sohamakl, 2019-12-06:
> Hi @sohamakl . I have the very same display and encounter some really strange issues with the wifi.. If I connect my RPi 3A+ to the power output of the display, I can not set up stable wifi [...]

Do you know how to configure this display to adjust the backlight by command or if you can hide the message "no signal" when hdmi is turned off?

Trenar, 2019-12-06:
No, not yet. Since I have more urgent problems (see #43), I had no time for this, yet. Ultimately, if there is no software solution, one could solder a wire between the hardware button and a gpio pin. Then you only need to programmatically set the gpio pin to logical high to turn on/off the display

Trenar, 2019-12-17:
sohamakl : in the meantime I successfully implemented a hardware solution for the screen switch. I use an optocoupler, connected to gpio pin 1 and ground on the one hand and to the two sides of the display's power button on the other side. In schedules.js I call a custom bash script that sets that pin on high for 50 ms and then back low again. Works very well :)

sohamakl, 2019-12-29:
> @sohamakl : in the meantime I successfully implemented a hardware solution for the screen switch. I use an optocoupler, connected to gpio pin 1 and ground on the one hand and to the two sides of the [...]

Trenar, this sounds good. Could you provide the bash script, new schedules.js and type of optocoupler you have used?

Trenar, 2019-12-30:
Hi,
I use the Kingbright KB817-M optocoupler and a small resistor between gpio pin 1 and it's input port.
You can find the scripts in my Teleframe fork. Have fun :)

sohamakl, 2020-01-03:
Thanks, it's working very well for me now.

closed 2020-02-13 by gegu