LukeSkywalker92/TeleFrame #9: Schwarzer Bildschrim mit TeleFrame Logo

carver79, 2018-11-16:
Hi Luke,

bin auf Dein Projekt gestossen und möchte es direkt umsetzen. Wenn auch mit begrenzten Kenntnissen. Ich habe einen Pi3B mit 2018-10-09-raspbian-stretch aufgesetzt und deinen one-line-installer verwendet. 
Fährt hoch, startet TeleFrame aber danach habe ich nur den schwarzen Bildschirm mit Logo in der mitte. Bot habe ich aufgesetzt, aber es schein nichts am Pi anzukommen. Den Token habe ich generiert und auch bei der Installation eingegeben. Zum test auch einen neuen Token generiert und per ssh upgedated. Aber ich stecke fest und weiss nicht wo ich ansetzten muss.
Liegt es an der config von TeleFrame oder am Bot...
Den Bot habe ich nur angelegt - kann ihm am iPhone jedenfalls mal nachrichten schicken :)

Hast Du eine Idee wo man hier angreifen kann?

Grüße
carver

LukeSkywalker92, 2018-11-17:
Und zeig mir bitte mal deine config. (Aber nehm vorher den bot Token raus ;))

LukeSkywalker92, 2018-11-17:
Also ich glaube die Fehler sind nicht so wild, aber sicher bin ich mir nicht.
Es gibt jetzt ein paar Sachen, die du versuchen kannst.

Als erstes würde ich vorschlagen du schickst mal ein Bild an den bot und schaust dann nochmal die logs an und zeigst sie mir, aber mit ein paar mehr Zeilen (`pm2 logs TeleFrame --lines 30`).

Hast du mit `sudo apt update && sudo apt upgrade` das system auf den neusten stand gebracht? wenn nicht, mach das mal und Versuchs nochmal.

Als letztes würde ich vorschlagen setzt du den pi nochmal neu auf. Also SD karte formatieren -> raspbian draufspielen -> system mit update und upgrade auf den neusten stand bringen und dann nochmal das install script ausführen.

Wir werden das problem schon finden ;)

LukeSkywalker92, 2018-11-17:
Kannst du bitte auch noch den error log zeigen?

carver79, 2018-11-18:
Sehr gern:
_(node:5007) UnhandledPromiseRejectionWarning: Error: 404: Not Found
    at buildConfig.then.then.then.then (/home/pi/TeleFrame/node_modules/telegraf/core/network/client.js:262:17)
    at process._tickCallback (internal/process/next_tick.js:68:7)
(node:5007) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:5007) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
ATTENTION: default value of option force_s3tc_enable overridden by environment.
ATTENTION: default value of option force_s3tc_enable overridden by environment.
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
ATTENTION: default value of option force_s3tc_enable overridden by environment.
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
ATTENTION: default value of option force_s3tc_enable overridden by environment.
[4359:1117/152938.472620:FATAL:bus.cc(1151)] D-Bus connection was disconnected. Aborting.
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
ATTENTION: default value of option force_s3tc_enable overridden by environment.
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
ATTENTION: default value of option force_s3tc_enable overridden by environment._

LukeSkywalker92, 2018-11-18:
Ok für mich schaut das so aus, als würde da etwas mit dem bot Token schief gehen.

Kannst du mal mit deinem browser den link `https://api.telegram.org/botXXX/getMe` aufrufen und XXX mit dem Token ersetzen?

Wenn da `{"ok":false,"error_code":404,"description":"Not Found"}` rauskommt, dann stimmt was mit dem Token nicht.

LukeSkywalker92, 2018-11-18:
ok die neue version ist hochgeladen. Wenn du im telegram Ordner `git pull` ausführst und dann `pm2 restart TeleFrame` eingibst sollte es funktionieren.

carver79, 2018-11-18:
Habe leider schon angefangen alles neu einzurichten - nur um sicher zu gehen.
Aber führe alles aus und bin gespannt -  gebe bescheid!
&Danke

carver79, 2018-11-18:
Perfekt :) - läuft.
Werds heute Abend mal genauer testen.
Noch eine Frage: welchen monitor hast Du verwendet?
Grüße!

LukeSkywalker92, 2018-11-18:
Ich hab mich für den Bildschirm entschieden: [https://www.waveshare.com/7inch-lcd-for-pi.htm](https://www.waveshare.com/7inch-lcd-for-pi.htm)

Den kann man aber leider nicht über den pi an und ausschalten. Dafür ist er günstig.

closed 2018-12-03 by LukeSkywalker92