LukeSkywalker92/TeleFrame #11: Zusätzliche Features

TBR-BRD, 2019-01-22:
Ist es möglich weitere Infos einzublenden? Interessant wären Uhrzeit, Datum und Wetter.

LukeSkywalker92, 2019-01-24:
Uhrzeit und datum sollte kein Problem sein. Muss halt eingebaut werden =)

Wie genau stellst du dir das mit dem Wetter vor?

LukeSkywalker92, 2019-01-31:
Puh. Das ist schon möglich, aber ich glaub das passt nicht wirklich zu der Anwendung.
Wenn man so viele features will, dann ist doch der MagicMirror genau das richtige oder?

TBR-BRD, 2019-02-05:
Das ist eine sehr gute Idee mit der Integration als Modul in den MM, das wäre doch genial wenn die TG Bilder im Hintergrund wechseln würden. Ich habe leider noch kein MM Modul programmiert. Ich weiss nur dass es ein Fullscreen Modul gibt, vielleicht könnte man die Funktionen koppeln? https://github.com/darickc/MMM-BackgroundSlideshow

TBR-BRD, 2019-02-19:
Ich hätte noch eine Frage dazu: kann man den TeleFrame automatisch ohne GUI laufen lassen? Er sollte dann nur die Bilder annehmen und dann im Verzeichnis "imageFolder" ablegen. Im Modul https://github.com/darickc/MMM-BackgroundSlideshow#using-the-module würde ich den Pfad dann dorthin legen. Würde das funktionieren?

LukeSkywalker92, 2019-02-21:
Im Moment noch nicht, aber das ist eine gute Idee. Wenn ich Zeit finde, kann ich das mal angehen.

LukeSkywalker92, 2019-02-21:
Habs eingebaut.  In der README findest du die Anleitung dazu.

closed 2019-02-21 by LukeSkywalker92

TBR-BRD, 2019-02-22:
Wow das ging schnell, vielen Dank. Ich wollte jetzt den Autostart im Fullscreen rausnehmen, habe gesehen du nutzt pm2, da habe ich überhaupt keine Erfahrung. Wie kann ich den Autostart von der Fullscreen Variante auf npm run botonly ändern?
Viele Grüße

LukeSkywalker92, 2019-02-22:
Du erstellst einfach eine Datei `irgendwas.sh` mit dem Inhalt 
``` bash
cd ~/TeleFrame
node run botonly 
```

Und tippst in der kommandozeile 
``` bash
pm2 delete 0
pm2 start irgendwas.sh
pm2 save
```
ein.

TBR-BRD, 2019-02-22:
Sorry ich bekomme es nicht hin.

pm2 show 0

status            │ errored                              │
│ name              │ botonly                              │
│ version           │ 1.2.0                                │
│ restarts          │ 15                                   │
│ uptime            │ 0                                    │
│ script path       │ /home/pi/TeleFrame/botonly.sh        │
│ script args       │ N/A                                  │
│ error log path    │ /home/pi/.pm2/logs/botonly-error.log │
│ out log path      │ /home/pi/.pm2/logs/botonly-out.log   │
│ pid path          │ /home/pi/.pm2/pids/botonly-0.pid     │
│ interpreter       │ bash                                 │
│ interpreter args  │ N/A                                  │
│ script id         │ 0                                    │
│ exec cwd          │ /home/pi                             │
│ exec mode         │ fork_mode                            │
│ node.js version   │ N/A                                  │
│ node env          │ N/A                                  │
│ watch & reload    │ ✘                                    │
│ unstable restarts │ 0                                    │
│ created at        │ N/A                                  │

Ich habe erstellt:

sudo nano /home/pi/TeleFrame/botonly.sh

cd ~/TeleFrame
node run botonly

sudo chmod +x /home/pi/TeleFrame/botonly.sh

pm2 delete 0

pm2 start /home/pi/TeleFrame/botonly.sh

pm2 save

sudo shutdown -r 0

… und es läuft leider nicht, bei status steht immer errored. Noch eine Frage, der MM wird doch auch per pm2 gestartet, geht das parallel? Der will nämlich auch nicht mehr.

Danke für deine Hilfe.

LukeSkywalker92, 2019-02-22:
Zeig mir bitte mal den Output von `pm2 logs 0`

TBR-BRD, 2019-02-22:
pm2 logs 0
[TAILING] Tailing last 15 lines for [0] process (change the value with --lines option)
/home/pi/.pm2/logs/botonly-out.log last 15 lines:
/home/pi/.pm2/logs/botonly-error.log last 15 lines:
0|botonly  |     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
0|botonly  |     at Function.Module._load (internal/modules/cjs/loader.js:507:25)
0|botonly  |     at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
0|botonly  |     at startup (internal/bootstrap/node.js:283:19)
0|botonly  |     at bootstrapNodeJSCore (internal/bootstrap/node.js:743:3)
0|botonly  | internal/modules/cjs/loader.js:583
0|botonly  |     throw err;
0|botonly  |     ^
0|botonly  |
0|botonly  | Error: Cannot find module '/home/pi/TeleFrame/run'
0|botonly  |     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
0|botonly  |     at Function.Module._load (internal/modules/cjs/loader.js:507:25)
0|botonly  |     at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
0|botonly  |     at startup (internal/bootstrap/node.js:283:19)
0|botonly  |     at bootstrapNodeJSCore (internal/bootstrap/node.js:743:3)

> Am 22.02.2019 um 17:40 schrieb LukeSkywalker92 <notifications@github.com>:
> 
> pm2 logs 0

LukeSkywalker92, 2019-02-22:
Ach ich bin ein blödmann.
In das Script muss `npm run botonly` und nicht `node run botonly`

TBR-BRD, 2019-02-22:
Ganz herzlichen Dank, es funktioniert!!!
Einen schönen Abend...

ingo46, 2019-11-08:
Nach jetzt fast vierwöchigen Betrieb bei meiner "Schwiegermutter" haben sich Bilder angesammelt, die ich gezielt löschen möchte. Wie lässt sich eine solche Funktion nachrüsten?