LukeSkywalker92/TeleFrame #25: Frameonly mode

s00500, 2019-09-24:
I have run into a stupid issue when the frame is actively turned on and off. when I send a picture and it takes more than 24 hours for the frame to start then telegram drops the cached image. To solve this I want to have a **frameonly** mode and a bot only mode. The two instances are running and are synced via Google Drive or some other could storage.

This way it is also easy to delete photos

Things that need to be done to do so:

Separate metadata into multiple files, one per photo
Create a frame only mode
Actively watch filesystem for changes
...

I already started the implementation of this :-)

Let me know if you have any additional ideas or thoughts

LukeSkywalker92, 2019-09-26:
Hi, sounds good.

Do you plan to run the botonly and frameonly mode on separate machines?

s00500, 2019-09-26:
Exactly, Ideally you could even have multiple frames for the same bot

kris711, 2019-12-26:
any news for this development? I'd really like to have a frameonly mode too 👍

s00500, 2019-12-28:
hm currently this is implemented on my own fork and it is working... but it is also not ideal and kills a lot of the other features... also there have been many changes in the meanwhile on the master branch...

sohamakl, 2020-01-25:
Are you still working on this? I could use the feature as well.

s00500, 2020-01-27:
No change since my last comment, checking my code and adapting it to the current master is very welcome, but currently I don't really have the time for it... sorry

gegu, 2020-02-13:
If someone is still interested and supports me a bit, I would start the development of an addon for this topic.