LukeSkywalker92/TeleFrame #26: system freeze

parachutesj, 2019-10-23:
Have a Pi3, almost every day system freezes, black screen not able to connect via SSH. However let a terminal open and found out, that it only becomes really really slow. 
Getting these errors in log: (id/token changed)

0|TeleFrame  | Failed to fetch updates. Waiting: 1s request to https://api.telegram.org/botabc:xyz/getUpdates?offset=0&limit=100&timeout=30 failed, reason: write EPIPE
0|TeleFrame  | Failed to fetch updates. Waiting: 1s request to https://api.telegram.org/botabc:xyz/getUpdates?offset=0&limit=100&timeout=30 failed, reason: Client network socket disconnected before secure TLS connection was established
0|TeleFrame  | Failed to fetch updates. Waiting: 1s request to https://api.telegram.org/botabc:xyz/getUpdates?offset=0&limit=100&timeout=30 failed, reason: Client network socket disconnected before secure TLS connection was established

what I noted, SWAP was completely full during the time.

Any hints what went wrong?

LukeSkywalker92, 2019-10-24:
I do not have a specific idea at the moment. But I have a question: Do you have the same bot running somewhere else? E.g. a browser or another teleframe? Or did you run it a short time before somewhere else?

parachutesj, 2019-10-24:
No, only this instance is running. Since SWAP was disabled it works. Will let it run for another 24h and then enable SWAP again to see what happens.

LukeSkywalker92, 2019-11-12:
Are there any news about your problem?

parachutesj, 2019-11-12:
not really. Since swap was deactivated, no more issues with frozen screen or slow system. But after 2-3 days, new images are not displayed anymore. So I send a picture, nothing happens. then a 2nd one and than both are going to be displayed in the screenshow. So longtime experience is rather unstable

LukeSkywalker92, 2019-11-20:
This sounds like something is wrong with the telegram bot. Since i did not hear something similar from others, i'm not sure if this is a problem of the teleframe software or something else. If you find out more, please tell me.

Worren, 2019-11-20:
I use TeleFrame with an raspberry 3A+ with 512MB RAM and experience the same issue - system will slow down after about a day of use. SWAP is full.

parachutesj, 2019-11-21:
Why not disable it?
sudo swapoff -a

LukeSkywalker92, 2019-11-21:
Ok this is really strange. At the moment I don't have any idea how the bot should fill the swap. Did you send many pictures/videos to the bot in that time? Or does this also happen, when the bot is "idle"? Maybe you can try to monitor if your used swap increases when you send something to the bot. It can also help to share your config file with us (sure without the bot token).

parachutesj, 2019-11-21:
I did send 15-20 pictures and 3 videos and then send every once and then another picture. I wanted to make sure it is running stable before getting it to the grandparents. Unfortunately it does not

LukeSkywalker92, 2019-11-21:
ok. could you please check if the swap gets also filled when you don't send something to the bot?

nautical-miles, 2019-11-21:
Hi,
I set up two frames and both freeze after a while. A reboot solved the problem for some time but it freezes again the next day. I now turned off SWAP for testing as parachutesj said. Let's see what happens.

Worren, 2019-11-21:
With increased swap size I got no slowdown after a day. I didn't send a picture during the timeperiod, but the used swap space increased about 50 MB in 14h.
I tried to disable swap once, but the Teleframe didn't boot anymore -  i think this is because of the 512mb of the A+.

Worren, 2019-11-21:
Config file:
var config = {
  botToken: 'XXX',
  whitelistChats: XXX,
  playSoundOnRecieve: "sound2.mp3",
  showVideos: true,
  playVideoAudio: false,
  imageFolder: "images",
  fullscreen: true,
  fadeTime: 1500,
  interval: 10 * 1000,
  imageCount: 30,
  newPhotoMessage: "Neues Foto von",
  newVideoMessage: "Neues Video von",
  showSender: false,
  showCaption: true,
  toggleMonitor: true,
  turnOnHour: 9,
  turnOffHour: 22,
  keys: {
    next: "right",
    previous: "left",
    play: "l",
    pause: "k"
  },
  voiceReply: {
    key: "a",
    maxRecordTime: 60*1000,
    recordingMessageTitle: "Voice Message",
    recordingPreMessage: "Recording for",
    recordingPostMessage: "in progress...",
    recordingDone: "Voice message sent sucessfully!",
    recordingError: "Voice message has failed!"
  }
};
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {
  module.exports = config;
}

LukeSkywalker92, 2019-11-21:
i analysed the memory usage a little and I have one idea what could cause the problem. Since TeleFrame is running inside a Browser window, every image and - more important - video is loaded from the browser into the memory. with lots of videos, this can be above 512 mb very easily. When I look at the dev tools of the chromium browser, it seems like all images and videos are kept in memory... I'm not sure if it is possible to tell electron to release the cached files after they where shown.. but at least we have a starting point...

Worren, 2019-11-21:
In this case it would help to stop Teleframe and restart it again? What are the commands? I would like to try it.

LukeSkywalker92, 2019-11-21:
pm2 restart TeleFrame

Worren, 2019-11-21:
Restart command reset the used swap space

LukeSkywalker92, 2019-11-22:
Small update: The increasing memory usage definitely happens in the browser window. I'm not sure if its only the videos or also the images. But a refresh of the browser window already helps to free the memory. I now try to find the reason for this..

LukeSkywalker92, 2019-11-22:
Ok I think I found and fixed the problem.
First the Problem:
I sent 4 images to the Bot and just let it run, while I monitored the performance. This was the result:

[image]

As you see, the number of Nodes and Listeners is increasing every time the shown image changes. This explains why the pi goes out of ram at some point and starts to swap.
The reason in the end were the img.onload() and video.onloadeddata() listeners, that were not removed from the images/videos after they were removed from the DOM. So there were still links to the media files and the garbage collection did not released them from memory.

The workaround I found was to set the onload functions and the media file elements to `null` after they were loaded. Additionally a `cleanCache()` of the browser was necessary. This is how the performance looks after these changes:

[image]
 
I implemented these changes in branch [hotfix/2.0.3](https://github.com/LukeSkywalker92/TeleFrame/tree/hotfix/2.0.3). Please try this branch on your pi and tell me if it fixes your issues. If it does, we can merge it to the master branch.

Worren, 2019-11-23:
I think you found the leak. For me it is now stable for 9 hours.
BTW: How do you delete images? Delete it from the images folder?

gegu, 2019-11-24:
Hi everbody!

IMO the problem have another background. Sometimes the `currentImage` could'nt be removed because a DOMeException occurs - . 
Frequently using move next or back reproduces the problem sometimes.

[Created a pull request for a hotfix](https://github.com/LukeSkywalker92/TeleFrame/pull/41)

LukeSkywalker92, 2019-11-24:
Hey @gegu. I think it's a combination of both. I suggest to implement both fixes. What do you think about it?

LukeSkywalker92, 2019-11-24:
I tried your hotfix and the number of elements is still increasing:

[image]

But I am aware of the problem you are describing, so thanks for the fix. I'm going to merge both into master now.

LukeSkywalker92, 2019-11-24:
closed by #45 and #42

closed 2019-11-24 by LukeSkywalker92

gegu, 2019-11-25:
@LukeSkywalker92 yes, i'ts a good idea. But I'm afraid the problem is deeper. I'll take another look at it in a quiet minute.