LukeSkywalker92/TeleFrame #34: Establish Git-Flow

LuHKae, 2019-11-20:
I would suggest to use a GIT workflow like gitflow.

My suggestion ist gitflow:
* A develop branch is created from master
* A release branch is created from develop
* Feature branches are created from develop
* When a feature is complete it is merged into the develop branch
* When the release branch is done it is merged into develop and master
* If an issue in master is detected a hotfix branch is created from master
* Once the hotfix is complete it is merged to both develop and master

Here is on graphical overview of the flow between the branches:
https://danielkummer.github.io/git-flow-cheatsheet/

mistau, 2019-11-20:
this makes perfect sense for me, I used the branch naming convention feature/ and hotfix/ already but we should agree on a full flow.

LukeSkywalker92, 2019-11-20:
👍  Very good idea. Since I was working alone on this project I did not think about something like this. But it seems like you guys are motivated to push this project further 😃 So it definitely makes sense to define a workflow. I like the idea of gitflow. This sounds promising. But we should add some development/contribution infos to the readme, if others want to contribute as well...

LukeSkywalker92, 2019-11-20:
Ok so I created the develop branch (again 😃). I think we should agree on the default settings go gitflow. So we use `git flow init -d` to initialise gitflow.

LukeSkywalker92, 2019-11-24:
I think we can close this issue, right?

closed 2019-11-24 by LukeSkywalker92