LukeSkywalker92/TeleFrame #38: Picture Freeze

nautical-miles, 2019-11-22:
Hi,
I create a new issue here as it seems my problem is different to the system freeze issue.
TeleFrame stops randomly on one picture and does not react on touching the screen. However logging on to the raspberry via ssh is possible. Also TeleFrame is turning off the screen at the correct hour when set up in the config. It also turns on the screen the next morning just to see the picture from last night sitting there for hours. Using pm2 restart TeleFrame does not help. Rebooting the raspberry makes TeleFrame work again for some time.

LukeSkywalker92, 2019-11-22:
did you have a look at the logs?

nautical-miles, 2019-11-22:
Does TeleFrame have it's own logs or is it stored somewhere else?

LukeSkywalker92, 2019-11-22:
just type pm2 logs TeleFrame

nautical-miles, 2019-11-22:
There is an error in the log: Failed to process updates. TelegramError Conflict terminated by other getUpdates Request...
I assume that running two screens with one bot is not working.

LukeSkywalker92, 2019-11-22:
Ok so we found your problem =) You are right. You need one bot per screen. Telegram does not allow to run multiple instances of one bot.

closed 2019-11-22 by LukeSkywalker92