LukeSkywalker92/TeleFrame #39: Nodejs Version 10x wird nicht unterstützt.

Mai1986, 2019-11-23:
Hallo Luke,

ich habe versucht die automatische Installation für die TeleFrame Software durchzuführen. Leider bekomme ich den Fehler das die Version 10 von Nodejs nicht unterstützt wird. Danach bricht die Installation ab. Nach ein bisschen Stöber habe ich es geschafft die Version 8 von Nodejs installieren. Der Fehler mit der nicht unterstützten Version ist weg. Es konnte dann aber leider nicht das Paket Elektron installiert werden. Ich bin leider echt ratlos aktuell. Liegt auch daran das ich vollkommener Neuling auf GitHub und in Sachen Linux bin. Ich hoffe du kannst mir da weiterhelfen?

VG 
Christian

sohamakl, 2019-11-23:
Hallo,

mit folgendem Kommando kannst du schauen welche Version von npm installiert installiert ist:

npm -v

Möchtest du sie aktualisieren geht dies wie folgt:

sudo npm install npm@latest -g
sudo reboot

Danach läuft alles wie gewünscht.

Mai1986, 2019-11-24:
Hallo,

danke erstmal für die Rückmeldung. Mein Problem ist, ich habe die aktuellste Version 10.x........

Beim automatischen installieren der TeleFrame Software bricht der Installer ab weil die Version 10 nicht von dem TeleFrame Paket unterstützt wird. Da komme ich im Moment nicht weiter. 
Besser kann ich es leider nicht beschreiben.

LukeSkywalker92, 2019-11-24:
Kannst du uns bitte noch ein paar Infos zukommen lassen:
* Welchen Raspberry verwendest du?
* Welches Raspbian hast du installiert?
* Wie bist du bei der Installation genau vorgegangen?

Mai1986, 2019-11-25:
Ich verwende einen PI 3B+ mit dem System Raspbian Buster Version 10.

Zu erst habe ich den PI für mein Netzwerk konfiguriert.
Danach habe ich mich per SSH eingeloggt. 
Dann habe ich gleich mit dem automatischen Installationspaket begonnen.

Ich habe mal den Terminalverlauf (Falls das hilft) kopiert und als Txt. hinzugefügt.

Ach ja, die Version von NPM ist 5.8.0

Brauchst du noch mehr Infos?
[Terminalverlauf.txt](https://github.com/LukeSkywalker92/TeleFrame/files/3888173/Terminalverlauf.txt)

sohamakl, 2019-11-25:
Versuch doch mal mit:

```
sudo apt-get update
sudo apt-get upgrade
```
und anschließendem 

```
sudo npm install npm@latest -g
sudo reboot
```

zu aktualisieren. Bei mit funktioniert es auf diese Weise.

LukeSkywalker92, 2019-11-26:
Funktioniert es mit dem Vorschlag von @sohamakl jetzt?

Mai1986, 2019-11-26:
Super, vielen vielen vielen Dank!  Es lag an der Version von npm. Nach dem Update auf die neueste Version hat die Installation funktioniert und der TeleFrame läuft :) Ich freu mich echt wie total. Jetzt gibt's doch noch ein Weihnachtsgeschenk für Oma und Opa.

LukeSkywalker92, 2019-11-26:
Sehr schön. Dann hoffen wir mal, dass das Geschenk gut ankommt :)

closed 2019-11-26 by LukeSkywalker92