LukeSkywalker92/TeleFrame #43: NPM error, Picture freeze

nautical-miles, 2019-11-24:
Hi,
my TeleFrame still freezes after some time. Touch does not react in any way at that point. SSH into the raspberry is fine. When looking into the log I see: "Draw call returned Invalid argument.  Expect corruption".
To be on the save side I did a fresh installation of raspbian and TeleFrame.

LukeSkywalker92, 2019-11-24:
Can you please attach the complete logs? And please add some infos about your hardware.

nautical-miles, 2019-11-24:
Here are the both log files I could find.
[TeleFrame-out.log](https://github.com/LukeSkywalker92/TeleFrame/files/3883902/TeleFrame-out.log)
[TeleFrame-error.log](https://github.com/LukeSkywalker92/TeleFrame/files/3883903/TeleFrame-error.log)

LukeSkywalker92, 2019-11-26:
can you please checkout #46 and see if it is the same? Some googling makes me think that it can be a problem with chromium and the raspberry. #46 uses a new electron(chromium) version so maybe it helps.

nautical-miles, 2019-11-27:
I still can't get it installed:
[image]

nautical-miles, 2019-11-27:
Got it finally installed. See #44 
Now it is testing if it still hangs after a while.

nautical-miles, 2019-11-29:
The new version is not stable on a raspberry pi 3. It still hangs after some time.
Is there a way to see if electron fails and restart it. It could be a workaround till the actual issue is solved.

nautical-miles, 2019-12-02:
For now I am restarting the Pi every hour. It is not nice but at least it does not sit there for days showing one and the same picture. I prefer a restart of electron frequently but could not find a way to do so. Doing a pm2 restart TeleFrame does not help.

sohamakl, 2019-12-03:
I have build three TeleFrames, two with Raspberry 3 and feature/touch-control branch with the freeze issue and one with raspbarry pi 4 that‘s running stable.

nautical-miles, 2019-12-04:
I turned off hardware acceleration for TeleFrame by adding:
app.disableHardwareAcceleration();
to main.js. As I do not send videos it doesn't really matter to me. TeleFrame is running since this morning 8:00 without any problems.

gegu, 2019-12-04:
That's interesting. I can't figure out the freeze on two pi3. One of them's been running for several days.
So far I have only tested with pictures, but have now sent a few videos.

But I disabled `toggleMonitor`. My screen gets into trouble when trying to turn off/on every second. That's why I asked for `toggleMonitor`.

sohamakl, 2019-12-04:
I‘m using fotos only utill now.

nautical-miles, 2019-12-05:
I'm also only using photos on my screens. This is a minimal setup without speaker and microphone.
@gegu: have a look at schedules.js and replace the first two asterisks in the create schedule command by numbers. That will stop Teleframe sending turn on/off commands every second for a full hour to your screens.

gegu, 2019-12-05:
Yes, but I have another idea. It might be better to check if the screen is turned off/on before issuing commands. I'm still looking at that.

sohamakl, 2019-12-06:
Yesterday I updated to npm 6.13.2 and nodejs 13.3.0 for testing and deleted all logs.

Two out of three TeleFrames are still running this morning. One is frozen again. Only entry in TeleFrame-error.log.

`Draw call returned Nicht genügend Hauptspeicher verfügbar.  Expect corruption.`

Trenar, 2019-12-06:
Can someone tell me about his/her installation procedure? I either get problems with nodejs 10.x or have a really bad graphical performance (laggy slideshow) using node 8.x...

nautical-miles, 2019-12-06:
> Yes, but I have another idea. It might be better to check if the screen is turned off/on before issuing commands. I'm still looking at that.

That's the same error I get even without your upgrade. One of my screens is still running after I turnede off hardware acceleration. The other one got stuck. Sadly without any error message in the log. A restart of TeleFrame with pm2 restart TeleFrame did not help.

sohamakl, 2019-12-06:
> Can someone tell me about his/her installation procedure? I either get problems with nodejs 10.x or have a really bad graphical performance (laggy slideshow) using node 8.x...

I did it this way:

##### node js aktualisieren 13.3.0

```
curl -sL https://deb.nodesource.com/setup_13.x | sudo -E bash -
sudo apt install -y nodejs
```

##### npm aktualisieren 6.13.2

```
sudo npm install npm@latest -g
reboot
npm -v
```

gegu, 2019-12-06:
I have node 10.x., installed, because this is the supported version (see [Node.JS distributions](https://github.com/nodesource/distributions/blob/master/README.md) ->Supported Debian versions ) and `npm 6.13.2` like you.

nautical-miles, 2019-12-09:
Installed the latest version of electron (7.1.2) after getting a hint that I had a typo in my command. It didn't take long till it freezes. Again the error: "Draw call returned Nicht genügend Hauptspeicher verfügbar. Expect corruption."
That might also be the reason why a Raspberry 4 does not have this error as it has more memory available.

gegu, 2019-12-11:
Have you set a high value for the `imageCount` configuration option?
I can't reproduce the problem on a pi3 using the default of `imageCount: 30` .
The memory consumption is approximately 350mb on a clean raspbian installation.

sohamakl, 2019-12-11:
I have an image count of 90 configured.

nautical-miles, 2019-12-12:
> Have you set a high value for the `imageCount` configuration option?
> I can't reproduce the problem on a pi3 using the default of `imageCount: 30` .
> The memory consumption is approximately 350mb on a clean raspbian installation.

I left the setting as is.

gegu, 2019-12-12:
Last night I also had the problem with `imageCount: 90` on one of my pi3. But I don't think it's because of the number of images to display.

Does `/var/log/messages` contain similar entries about mmc_rescan?
```
Dec 12 04:37:54 raspberrypi kernel: [28752.088460] Workqueue: events_freezable mmc_rescan
```
followed by some exception information.

nautical-miles, 2019-12-15:
I could not find anything like this in /var/log/messages on my screens.

kris711, 2020-07-05:
I'd like to confirm this problem too.
It seems to be related to the screen hardware, because same setup runs like a charm with RPI 7" Display. But using a HDMI Waveshare 10.1 screen i can reproduce the freeze by

- toggle on Screen by defined hour in ConfigScript
- just waiting some hours with disabled toggling

I am using the hdmi_default.js for screenConfig.

`pi@Frame:~/TeleFrame $ npm restart`

> teleframe@3.1.0 start /home/pi/TeleFrame
> electron .
> 
> 
> **(electron:5691): Gtk-WARNING **: 10:42:23.827: cannot open display**:
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! teleframe@3.1.0 start:  electron .
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the teleframe@3.1.0 start script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/pi/.npm/_logs/2020-07-05T08_42_23_903Z-debug.log

` /home/pi/.npm/_logs/2020-07-05T08_42_23_903Z-debug.log`

> 0 info it worked if it ends with ok
> 1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'restart' ]
> 2 info using npm@6.14.4
> 3 info using node@v10.21.0
> 4 verbose run-script [ 'prerestart',
> 4 verbose run-script   'prestop',
> 4 verbose run-script   'stop',
> 4 verbose run-script   'poststop',
> 4 verbose run-script   'restart',
> 4 verbose run-script   'prestart',
> 4 verbose run-script   'start',
> 4 verbose run-script   'poststart',
> 4 verbose run-script   'postrestart' ]
> 5 info lifecycle teleframe@3.1.0 prerestart: teleframe@3.1.0
> 6 info lifecycle teleframe@3.1.0 prestop: teleframe@3.1.0
> 7 info lifecycle teleframe@3.1.0 stop: teleframe@3.1.0
> 8 info lifecycle teleframe@3.1.0 poststop: teleframe@3.1.0
> 9 info lifecycle teleframe@3.1.0 restart: teleframe@3.1.0
> 10 info lifecycle teleframe@3.1.0 prestart: teleframe@3.1.0
> 11 info lifecycle teleframe@3.1.0 start: teleframe@3.1.0
> 12 verbose lifecycle teleframe@3.1.0 start: unsafe-perm in lifecycle true
> 13 verbose lifecycle teleframe@3.1.0 start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/TeleFrame/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
> 14 verbose lifecycle teleframe@3.1.0 start: CWD: /home/pi/TeleFrame
> 15 silly lifecycle teleframe@3.1.0 start: Args: [ '-c', 'electron .' ]
> 16 silly lifecycle teleframe@3.1.0 start: Returned: code: 1  signal: null
> 17 info lifecycle teleframe@3.1.0 start: Failed to exec start script
> 18 verbose stack Error: teleframe@3.1.0 start:  electron .
> 18 verbose stack Exit status 1
> 18 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
> 18 verbose stack     at EventEmitter.emit (events.js:198:13)
> 18 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
> 18 verbose stack     at ChildProcess.emit (events.js:198:13)
> 18 verbose stack     at maybeClose (internal/child_process.js:982:16)
> 18 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
> 19 verbose pkgid teleframe@3.1.0
> 20 verbose cwd /home/pi/TeleFrame
> 21 verbose Linux 4.19.118-v7+
> 22 verbose argv "/usr/bin/node" "/usr/bin/npm" "restart"
> 23 verbose node v10.21.0
> 24 verbose npm  v6.14.4
> 25 error code ELIFECYCLE
> 26 error errno 1
> 27 error teleframe@3.1.0 start:  electron .
> 27 error Exit status 1
> 28 error Failed at the teleframe@3.1.0 start script.
> 28 error This is probably not a problem with npm. There is likely additional logging output above.
> 29 verbose exit [ 1, true ]

`lightdm/x-0.log`
> Draw call returned Invalid argument.  Expect corruption.

`logs/TeleFrame-error.log`
> Draw call returned Nicht genügend Hauptspeicher verfügbar.  Expect corruption.

I can also confirm that this screen is blanking after some time. A touch command recativates it.

I suppose TeleFrame is adding 
`xserver-command=X -s 0 -dpms`
to lightdm config, but this is not enough i could solve this issue by adding

> `# Disable Screensaver`
> setterm -blank 0 -powerdown 0s

in `rc.local` file

However the Freeze behaviour shows no difference with or without this entry.

As described by @nautical-miles I tried to restart lightdm but this ends up in a blank screen without any session. 
At the moment only a reboot helps. Howerver terminal access is possible the whole time.

Now I am going to play with Video options in boot/config.txt but I would be glad for any ideas.

kris711, 2020-07-09:
just to update this issue: I could solve by changing hdmi options in boot/config.txt
As this is very hardware individual I will create a wiki page for my display (Waveshare 10.1inch HDMI LCD (B)).
However for me it is not a TeleFrame issue.

Smigstr, 2020-07-22:
I was having issues with my teleframe freezing every time when I was showing a particular 12 second video. If I removed every line talking about videos on config.json it begun working again.

Still having other freezing issues tho.

kris711, 2020-07-22:
maybe its a good idea to publish your [log files](https://github.com/LukeSkywalker92/TeleFrame/issues/43#issuecomment-653870643)? Perhabs someone can help then.

Smigstr, 2020-07-23:
I know, sadly I don't have the logs easily available right now:/ 
nor i have a raspi to recreate them around me.

It was quite recreatable tho, I kept using these as my test videos:
https://www.dropbox.com/sh/kl4vpdwrq0d8n4n/AACYdQnhgImRbSupbggayp24a?dl=0

I had everything else as example config, but I had added a reboot admin command. Also the raspbian image was completely stock, except I had ran updates to have the newest versions of everything.