LukeSkywalker92/TeleFrame #49: Admin Mode, annotations and fadeTime issue

Trenar, 2019-11-30:
Hi Luke, 
first of all thanks a lot for your awesome project! 

I have some ideas for improvements, but unfortunately no experience with telegram bots and javascript... 
1. I'd really love an admin-feature for rebooting / updating and so on. I just saw, that there is a PR in the pipeline.. Hope to see this feature being implemented soon :) 

2. Is it possible to have the message and sender name remain as long as the image is shown? By now, I can only see it for a few seconds and I suspect my grandparents to miss it...

3. When the image changes, the screen blinks white or black for a few milliseconds.. It is most likely related to the fading between images, since it does not occur if I set fadeTime to 0 in the config.js. However, with fadeTime=0, the screen turns black for a short period and then the next image is shown. Would be nice if the black period wasn't there. PS: I am using an RPi 3A+ connected to a display (my tv screen) via HDMI. 

Looking forward for new updates and features in this project :)

Trenar, 2019-12-03:
Some more thoughts, since I worked a little with the code in the meantime...

2. I think it makes sense to have a config parameter for the amount of time, the sender and caption is shown. Maybe there can be three options like 0,1,2 meaning not shown, full time and half time of the slideshow-interval. 
BTW: @LukeSkywalker92 , is there a fallback in the case that a user forgets one of the parameters in the config? Maybe we should implement that the TeleFrame uses the config.example.js values in case of undefined parameters. 
Related question: I once tried to "deactivate" the voice reply action by removing the parameters from the config file. However, this gave errors on long-press. Is there a proper way to deactivate this feature? 

3. I solved that. I had to uncomment the `config_hdmi_boost=4` line in `/boot/config.js`. Seems to be related to the RPi 3A+, if I'm the only one with this issue.

gegu, 2019-12-15:
Config parameter for the amount of time and #69 was implemented in [feature/caption-sender-fullscreen](https://github.com/LukeSkywalker92/TeleFrame/tree/feature/caption-sender-fullscreen).

config.js
```js
...
  interval: 10 * 1000,
  showCaption: true,
  // To output sender caption, use the entire screen
  useFullscreenForCaptionAndSender: true,
  // Defines the percentage of the duration of <interval> to show sender and caption.
  // minimum value: 10  = fade out after 10% of <interval>
  // maximum value: 100 = full time. sender and caption does not fade out
  senderAndCaptionDuration: 50,
...
```
But the release has to wait until we have #66 finished.

Trenar, 2019-12-17:
Nice implementation, thank you!

What is the status of #35 and #53? Also postponed until #66 is decided?

gegu, 2019-12-17:
IMO it is better to integrate the waiting PR's only after changing the configuration handling. Especially #35 will only work if the configuration is to be changed by external commands.

Trenar, 2019-12-17:
Sure, I agree. Too bad that Christmas is that early this year :P

gegu, 2019-12-17:
There are seven days left. I am optimistic that we will be able to do a few more things 😏

Trenar, 2020-01-02:
closed with PR #86 and #35

closed 2020-01-02 by Trenar