LukeSkywalker92/TeleFrame #54: Show URL‘s in TimeFrame

sohamakl, 2019-12-01:
In addition to photos and videos, what do you think about displaying URLs that are sent as links? So you could also use TimeFrame as a dashboard.

LukeSkywalker92, 2019-12-03:
Sounds like a nice Idea. Can you please describe the expected behavior a bit more detailed?

sohamakl, 2019-12-03:
If you could display web content as a full screen, this would be great in a first version.

No frames or controls should be displayed.

It would be really cool to be able to use the web page in pause mode, if the website offers dynamic content.

LukeSkywalker92, 2019-12-03:
But how do you resume the slideshow when controls are disabled?

sohamakl, 2019-12-03:
To pause you have to fade in the navigation bar. It could be static visible until you resume.

sohamakl, 2020-03-08:
Does anyone have time and desire to work on this topic?

Unfortunately I lack the experience to tackle it myself.

sohamakl, 2020-10-16:
I wanted to ask if there is someone who would implement it.

I can offer myself for testing.