LukeSkywalker92/TeleFrame #58: Switch off Raspberries LED's

sohamakl, 2019-12-06:
I wan't to suggest an option in the Installer script where the user can choose to switch off the Raspberry Pi power and status LED. I have tested with a Raspberry Pi 4. The following commands do the trick.

```
# led0 green, led1 red
sudo sh -c 'echo none > /sys/class/leds/led0/trigger'
sudo sh -c 'echo none > /sys/class/leds/led1/trigger' 
```

LukeSkywalker92, 2019-12-09:
sounds good. will you implement this?

gegu, 2019-12-11:
Unfortunately this does not work on a pi3 with current Debian buster.

Can you please check whether the following commands also lead to the desired result?

```
sudo sh -c 'echo 0 > /sys/class/leds/led0/brightness'
sudo sh -c 'echo 0 > /sys/class/leds/led1/brightness'
```
I think there should be a way to configure it, because it won't work on other platforms.

mistau, 2019-12-13:
checked to use the brightness as proposed by gegu, this works on my pi3b+ using Debian Buster as well.
Will provided an updated patch with config options as soon as we settled #66

mistau, 2020-01-02:
closed with #89

closed 2020-01-02 by mistau