LukeSkywalker92/TeleFrame #59: Aspect ratio correction

palmerich, 2019-12-06:
Hey,

I have a custom rapsi screen (with driver and all) and the ratio comes out slightly wrong... not much but a bit.

Is there an INI file or something where a correction could be set?

Thanks,
Andreas

LukeSkywalker92, 2019-12-12:
Is it only the teleframe software, that is shown with wrong ratio? Or also the raspbian stuff?

palmerich, 2019-12-12:
Not sure but don't think. Background and icons look ok. Driver inst went ok and the pixels are not blurry or anything (which I think they would be if the resolution is not okay). But since I do not use this device for anything else a hint how to change the overall ratio would be appreciated quite the same.

BR
Andreas