LukeSkywalker92/TeleFrame #63: Bot connection loss

Trenar, 2019-12-09:
Hi all,
I have still another problem with the TeleFrame. It seems that the bot looses the server-connection. The software itself works fine, no freezing system or so, but if I send an image, it will not be displayed until I restart the software (or pi). 
Can this be related to wifi power management in the pi or is it a software issue? By thinking over this, I thought about a notification (or a simple symbol like the pause-symbol) when the internet-connection is lost. I'm however not quite sure how to implement this...

palmerich, 2019-12-10:
just checking (noob too) - did you try with a wired connection to narrow it down?

Kind regards,
Andreas

gegu, 2019-12-11:
I tested this by triggering a reboot of my firewall. It then takes some time for the messages to arrive after the connection is reestablished - here about half an hour.

Trenar, 2019-12-12:
> did you try with a wired connection to narrow it down?

I'm running TeleFrame on a RasPi 3A+, which has no wired connector... 

> I tested this by triggering a reboot of my firewall. It then takes some time for the messages to arrive after the connection is reestablished - here about half an hour.

For me it seems like the reconnection can't be reestablished at all.. Tried running TeleFrame over night (without toggling monitor or so), sent new photos but they won't get displayed. Sending "Hi" to check if the bot is alive also gives no answer.

Trenar, 2019-12-13:
Today I had another issue (which was related to the router failing) but the bot could not reconnect after re-establishing network connection. When the pi was reconnected, I ssh'ed to it and opened the screen session where Teleframe is running in. There was this error message
<code>Failed to fetch updates. Waiting: 1s request to https://api.telegram.org/botTOKEN/getUpdates?offset=569122325&limit=100&timeout=30 failed, reason: read ETIMEDOUT</code>

This looks like it tries to connect a hundred times and then gives up. Do I get this right? How can we handle this? I was thinking of a daily reboot of the pi or so, or a regular check for a valid network connection..