LukeSkywalker92/TeleFrame #69: Text rendering on Screen width, not Picture width

chemical1979, 2019-12-14:
The family posts a lot of vertical taken pictures with some text along with it. This works pretty well in general, but I was wondering whether it is a big change to render the text (maybe also the name of the sender) on the screen width, instead within the actual picture.

Sometimes, if you receive a picture with some more words like e.g. "Look at him, holding his first bottle of milk" along with a picture taken in portrait mode, there is literally nothing left of the picture :-) 
The text fades out a few seconds later, which is great, but I was wondering if this could be improved by using the capacity of the actual screen instead.

Thank you!

gegu, 2019-12-14:
That should be possible. We'll take a look.

Trenar, 2019-12-16:
Related issue: on the development branch I have some problem with videos. They are (in landscape mode) not scale to screen width but only screen height. Hence, if taken in a wide screen aspect ratio, the left and right edges are cutted, a scroll bar is shown (pretty ugly) and the sender/capiton labels are cutted. Unfortunately I don't have much time to look into the code for the video rendering.. I hope it to be only one option in the renderer

gegu, 2019-12-17:
For videos the attributes `videoWidth`/`videoHeight`must be queried.

Fixed in feature/caption-sender-fullscreen - 7795020

Trenar, 2019-12-17:
Funny, I found this solution 2 hours ago and wanted to create a PR :D thanks a lot :)

gegu, 2020-01-07:
Done in develop

closed 2020-01-07 by gegu