LukeSkywalker92/TeleFrame #70: Sending commands to TeleFrame

nautical-miles, 2019-12-15:
Hi,
I wonder if it makes sense to send commands to the TeleFrame. Background why I am asking:
The TeleFrame I have made is in a case where you have no access to the buttons of the screen. It will not get a schedule for turning on and off as the usage times vary. Also the user will send pictures to the TeleFrame.
Long story short: I have modified the bot.js in this way that the screen can be turned on and off with a simple command.
I like to have your opinion to this modification.

Trenar, 2019-12-17:
Isn't that part of the admin mode in #35? I'd really love this feature, would be a great opportunity to run a teleframe at another place and have some kind of debugging tools.
Also discussed in #66 is the ability to modify the config parameters via messages

nautical-miles, 2019-12-17:
You are right. Closing this.

closed 2019-12-17 by nautical-miles