LukeSkywalker92/TeleFrame #71: Problems with audio: how to debug?

stefanoco, 2019-12-16:
I'm unable to make the audio generation work properly, and can't understand how to debug. Installed TeleFrame on a Raspberry using the provided script, auto-run with PM2. Is any log file available in this configuration?

LukeSkywalker92, 2019-12-16:
You can run `pm2 logs TeleFrame` to see the logs.

stefanoco, 2019-12-17:
Thanks, and in case someone else follows this discussion (with limited knowledge of PM2 as myself): if you installed TeleFrame with the provided script on a Raspberry the logs are reached with `pm2 logs TeleFrame` as `pi` user, not `root` or sudo.

LukeSkywalker92, 2019-12-17:
Could you fix your audio problems?

stefanoco, 2019-12-17:
Not yet I'm trying now

stefanoco, 2019-12-17:
...ok solved was as simple as an audio peripheral not being recognized/used by Alsa.
How to debug on a Raspberry, TeleFrame installed with the script:

* look at TeleFrame logs by `pm2 logs TeleFrame` (as `pi` user) while recording  an audio message (reveals problems with Alsa)
* look at system logs by `sudo tail -f /var/log/syslog` while unplugging/plugging an audio peripheral on USB (reveals problems recognizing the peripheral)
* google "raspberry usb audio ..." and the shown error

In my case in the end I solved by modifying two lines on `/usr/share/alsa/alsa.conf`:

```
defaults.ctl.card 1
defaults.pcm.card 1
```

closed 2019-12-17 by stefanoco