LukeSkywalker92/TeleFrame #72: Whitelist doesn't work in Groupie chats

leggler, 2019-12-16:
Hey everybody,
Ich am besten toll GitHub, so pleased Apologie If this ist not the right Place For my question.

Anyway, i added the bot to a group Chat. Said 'hi' to the bot, added the chat-id to the Whitelist (which Works fine with Personal Chats), but i only geht the following msg: 

"Hey there, this bot is whitelisted, pls add your chat id to the config file"

Only way ich can make the group Chat work is by disabeling whitelisting.

Am i Missing Something?

LukeSkywalker92, 2019-12-16:
Hey,

did you add a "-" before the group id?

leggler, 2019-12-16:
Yes! Works Like a Charm! Thanks For the hint, issued closed! ;) And in
General thanks For the GREAT Work!

Btw, one more question, very basic: how do i Stop teleframe once it has
Started (eg for configuration testing)? The development-frontend doesnt
Help me...

LukeSkywalker92, 2019-12-16:
Just use the pm2 commands:
```
pm2 start TeleFrame
pm2 stop TeleFrame
pm2 restart TeleFrame
```
But when you change something in the config, it should automatically restart.

closed 2019-12-16 by LukeSkywalker92