LukeSkywalker92/TeleFrame #73: Channel Support

koeberlue, 2019-12-16:
I added the TeleFrame bot as an admin to a channel. According to the telegram API it should be able to read all messages from channels, but images sent to the channel are not displayed on the TeleFrame. Is there any possibility to get channel messages working?

#### Use Case 1

Some family members also want to receive the images, but don't necessary need the TeleFrame. Add the same time I want to prevent them posting hundreds of pictures from their last vacation to the TeleFrame. So I created a channel where only some of us have admin rights and no one else should be able to post images.

#### Use Case 2

Some family members don't use telegram but upload their images to out Nextcloud instance. So I created a NodeRed flow which downloads images from specific Nextcloud folders and sends them to the channel, where the TeleFrame bot picks them up and displays them on the frame. Apparently this can not be done in a group, because bots can not see messages of other bots in groups, but only in channels.

---

By the way: Thanks for the great work. I really like this project and hope to contribute when I find some time to spare.

sohamakl, 2020-01-25:
I use a private group. It works without problems. Just don't forget to allow the bot to receive images with `/setprivacy disable`. If it works with channels I did not test.

koeberlue, 2020-01-29:
Thanks for the suggestion. I got the bot running in a private group successfully, too. But as explained in the use cases, I need it running in a channel which seems not to work.