LukeSkywalker92/TeleFrame #74: Whitelist dialogue

LukeSkywalker92, 2019-12-16:
Maybe we should show a dialogue when the bot receives a "hi" from a non whitelisted chat. This dialogue could show a "add 'sender name' to whitelist" button. This would be one more step towards a texteditor free configuration.
We should wait with this for #66.

Trenar, 2019-12-16:
This seems a bit unsafe to me.. What is the whitelist for, if anybody sending "hi" can add himself to it? IMHO, at least an admin of the bot would need to accept that user/chat-ID (asking for the admin mode #35 with some more improvements...).

gegu, 2020-02-13:
I think we could make that work.

But then we would also need a blacklist in which the rejected chatid's would be stored.

@Trenar, the sender would have to be on site so that he can confirm himself by touching  'Confirm' on the TeleFrame screen.

Trenar, 2020-02-13:
But if he isn't on site but requests access and my grandma presses okay, not understanding what's happening? I'd prefer the admin to actively whitelist chat ids

gegu, 2020-02-14:
Yes, I am familiar with the problem that technically less experienced people always like to confirm with "Yes I do". 🙈 
But this feature has to be enabled in the config anyway and should not be active by default.