LukeSkywalker92/TeleFrame #77: High Memory Consumption forces swapping on 512MB Pis

NDscham, 2019-12-18:
Hi everyone,
I attempted to get this running on a Pi 3A+ which has the same Quadcore CPU as the 3B, but only 512 MByte of RAM. This will result in swapping, flickering and occasional crashes. It might be worth looking into lowering memory demands and also eliminating unnessary processes.
I would go as far as of getting rid of the window manager as it is not really needed for this purpose.

What do you think?

Trenar, 2019-12-18:
Can you try increasing the swap (see e.g. https://wpitchoune.net/tricks/raspberry_pi3_increase_swap_size.html ) and see if the problem persists?

Trenar, 2019-12-18:
[image]

On my 3A+ I can not see a full swap memory. RAM is quite full, but not fully exploited. But as I said in #75 , I disabled the fkms OpenGL acceleration, which caused a lot of trouble for me.

closed 2020-02-13 by gegu