LukeSkywalker92/TeleFrame #79: Voice message breaks off after about 6 seconds

firebbbbb, 2019-12-20:
If I want to send a voice message, the recording stops after about 7 seconds and is sent. I know it shouldn't be silent for more than 2 seconds, but that also happens while I'm talking.

LukeSkywalker92, 2020-07-27:
Did you have a look at the logs? Can you share them with us?

KateiRen, 2020-07-28:
@firebbbbb: Can you hear your spoken words on the telegram client that is receiving the 7 seconds snippet? Is it very silent and noisy? I had the same when using the SunFounder USB 2.0 Mini Microphone.

Record level might be too low (use alsamixer or the graphical interface to increase).
The recording device might not be selected as default input (create a .asoundrc defining your config or use the graphical interface to select available devices).

@LukeSkywalker92: where can I find the log? Need to resolve some other issues on my end....

LukeSkywalker92, 2020-07-29:
if you are using pm2, the logs should be in `~/.pm2/logs`, or you can access them via `pm2 logs TeleFrame`.
if you are manually running TeleFrame, the logs are written to the terminal...