LukeSkywalker92/TeleFrame #81: System didn_t start - Black Screen with Logo

Elwetritsch71, 2019-12-23:
Hi there,

I have tried to set up the teleframe, but unfortunately it failed when I start the program. I have tried both versions, with pm2 and manually, it just freezes a black screen with the TeleFrame logo in the middle. The bot is recognized correctly and is active, the last entry is in the log:
info: Renderer started…

One other crazy thing, after the freeze, the keyboard no longer responds to the letters "A" + "L" and the arrows "left" and "right", if the program is canceled, it works again.

System:
Pi 3 B +
Raspian Buster 2019-09-26, all updates applied
Node V11.15.0
electron@7.1.7

Can anyone help me further?

Thanks and regards, Udo
-----------------------------------------------------------------------
Hallo,

ich habe nun auch versucht den Teleframe aufzusetzen, scheitere aber leider beim Programmstart. Ich habe beide Versionen, mit pm2 und manuell ausprobiert, es friert immer nur ein schwarzer Screen mit TeleFrame-Logo in der Mitte ein. Der Bot wird richtig erkannt und ist aktiv, im Log ist der letzte Eintrag
info: Renderer started …

Was mich zusätzlich stutzig macht ist, dass nach dem Freeze die Tastatur nicht mehr auf die Buchstaben "A" + "L" und die Pfeile "links" und "rechts" reagiert, wenn das Programm abgebrochen wird geht es wieder.

System:
Pi 3 B+
Raspian Buster 2019-09-26 , alle Aktualisierungen aufgespielt
Node V11.15.0
electron@7.1.7

Kann mir jemand weiterhelfen?

Danke und Gruß, Udo

Trenar, 2019-12-23:
Hi Udo,

there is nothing unexpected in what you describe. The black screen with logo is the default screen, when no images are sent to the bot yet. At startup (either manual in terminal or via pm2 using `pm2 logs TeleFrame`) you should see log message `Bot started!` followed by `Using bot with name <YOUR_BOT_NAME>`. That tells you that the bot connection is established correctly and you can send images to it. If the two above message (or at least the latter one) don't show up, you need to check your bot token in the config file.

The non-responsive keys are related to the navigation within the software. Arrow keys are used to navigate through the slideshow, <kbd>a</kbd> is used for voice message recording and <kbd>l</kbd> pauses the slideshow (or resumes). 

Hope that helps.

Elwetritsch71, 2019-12-23:
Hi Trenar,

thanks for the quick response! I'm glad that it doesn't seem to be the system. Do you have an idea which settings I missed at the bot? I can send messages and receive them on devices other than the teleframe.

Regards, Udo

Trenar, 2019-12-23:
Have you set the bot token in `config/config.js`?

Remember that each bot token can only be used once at a time. No multiple instances possible :) also, if you have your bot within a group, you have to change the privacy settings for your bot using the botfather. You may also need to check the whitelisted chats in your config

Elwetritsch71, 2019-12-23:
The bot token is set, xxx in my comment stands for the bot name.

I'm not used in the telegram bot, what do you mean with "once at a time"? The privacy mode is disabled, but I'm using the bot in our family group. The group setting within the bot is enabled, should I turn this off? Actually, there is no whitelist in use.

Trenar, 2019-12-23:
With "once at a time" I mean that you can't have two Teleframes running using the very same bot token.

Privacy mode off and group mode on is the correct setting. Try sending images to the bot directly, not in the family group. That way you can figure out if the problem is related to the group or not. Otherwise, I can't see what's wrong with your installation, sorry..

LoiX07, 2019-12-25:
Did the bot answer to hi and help ?

HaraldHopf, 2020-02-06:
Hallo zusammen,

ich habe ein ähnliches Problem.
Mein TeleFrame kommt nicht über "Black Screen with Logo" hinaus.
Mit dem Befehl "pm2 logs TeleFrame" kommt bei mir nicht die Medlung "[Main] info: Using bot with name xxx."

Es wäre super wenn mir jemand helfen könnte!
Danke und Gruß
Harald

Trenar, 2020-02-08:
Moin Harald,
Hast du den bot token korrekt (kein tippfehler oä) im config file angegeben?

HaraldHopf, 2020-02-09:
Hallo Trenar,

nachdem ich jetzt TeleFrame nochmals neu installiert habe und den bot token jetzt korrekt eingegeben habe funktioniert es endlich :-)
Vielen Dank für die Hilfe.
Harald

closed 2020-02-13 by gegu