LukeSkywalker92/TeleFrame #91: New touch command "Start from newest"

sohamakl, 2020-01-03:
What do you think about a new feature to restart the slideshow by a new touch command. So you can quickly see if new photos have arrived in your absence. Provided of course that the photos are sorted by Creation Date.

Trenar, 2020-01-03:
I don't think this is very useful for most users, since the frame is not accessible directly. At least I built three of them and distributed them among my family. 

Hence, I came up with another solution, which I want to create a PR for, when I find the time. Until now, it is just quick and dirty without config parameter or so..

In bot.js, in the function of `this.bot.on(['photo, 'video', 'document'], ...`, I check for the extension of the arrived file. If it is not one of the known formats (mp4, gif, png, jpg), I let the bot send a reply that the format is not supported. In case of the known formats, it replies with an emoji (" :+1: :movie_camera: " for Video, " :+1: :camera_flash: " for images). I find this very handy, since I get a direct reply on my sent image within telegram without doing anything. 

You can find it in my fork. Let me know what you think about this approach

sohamakl, 2020-01-03:
I think you got me wrong. I gave my grandma a TeleFrame. Now she's out all day while I send her new pictures. When she comes home in the evening she would like to see if new pictures arrived and jump directly to the new pictures instead of waiting until they are shown in the slideshow.

Trenar, 2020-01-03:
Indeed, I got you wrong. Now I like your idea :)

gegu, 2020-01-04:
We can implement that. But I would need a suitable CSS class definition from the pool of fontawesome for the symbol in the touchbar

sohamakl, 2020-01-04:
What do you think of these for one or more new photos arrived since last touch on symbole?

https://fontawesome.com/icons?d=gallery&q=photo

[image]

gegu, 2020-01-06:
It's fine for me.

gegu, 2020-01-23:
Closed by #101

closed 2020-01-23 by gegu