LukeSkywalker92/TeleFrame #93: New screen definition

sohamakl, 2020-01-03:
Sorry, I'm not getting along with github very well yet, so I don't want to make any pull requests or merges.

It would be great if you look at my suggestion and hopefully  include it in the project.

I took up Trenar's good idea and implemented the power on/off function for the Sunfounder HDMI display via optocoupler.

In this context the following 'files' were created:

- New screen definition file
- Shell Skript from Trenar adapted to my individual configuration
- Circuit diagram
- Board layout
- Datasheet optocoupler
- Two Photos

I also noticed that the wiringpi extension used in the script needs to be updated on the Raspberry Pi 4 first.

This may need to be included in the installer script and works as follows:

```
cd /tmp
wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb
```
[Screen definition Sunfounder 10.1 hdmi display.zip](https://github.com/LukeSkywalker92/TeleFrame/files/4020901/Screen.definition.Sunfounder.10.1.hdmi.display.zip)

Trenar, 2020-01-03:
Hi @sohamakl, nice to see you working on this. 
At first I want to mention #2 to let people see our recent discussion :) 

It may be confusing for people that you call pin 29 in the shell script but use 21 in circuit diagram (at least it was for me). So let me attach [this link](https://www.aranacorp.com/wp-content/uploads/raspberrypi-gpio-wiringpi-pinout.png) to clarify the pinout used by wiringpi / the gpio utility.

How do you feel about having the `gpio mode XX out` line not in the `screen_switch.sh` script but as the `cmdInit` of the screen config? However I feel like both solutions are not ideal, best would be an additional config parameter and a fallback pin number. 

Shouldn't the wiringpi core be update when running `sudo apt update && sudo apt upgrade`? I think this is done anyhow at some point of the installer script. 

Last comment: The image of the optocoupler wiring is very blurry. 

I could prepare a pull request once I find time for it.

sohamakl, 2020-01-03:
> Sorry, I'm not getting along with github very well yet, so I don't want to make any pull requests or merges.
>
> It would be great if you look at my suggestion and hopefully include it in the project. [...]

Hi @Trenar,

moving the gpio definition to the config file is a very good idea.

I will take some new and better photos when I back home on Sunday.

I had to update wiringpi manually. I also read on the developer's website that he does not want to support future versions of Raspberry Pi.

Trenar, 2020-01-03:
Hi,
I played around a little.. Moving the Pin param to the config is imho not feasible, since the config parameters are not accessible where the screen config file is loaded. Therefore, I added a variable in the screen config file which has to be set manually. For now I set it to your default pin 29. 

See PR #94 and please test it, just to be on the save side :)

sohamakl, 2020-01-04:
I took three more photos with bit better quality than before.

[image]
[image]
[image]

Trenar, 2020-01-04:
Looks good. However I think we don't need to upload these images to the git, but instead refer to this conversation at some point, maybe in the `screens/README.md` of #97

sohamakl, 2020-01-04:
Good idea, could you be so kind and add it to `screens/README.md`? I really need to learn how to use Github, sorry.

Trenar, 2020-01-04:
@gegu will do so :)

gegu, 2020-01-05:
I think this would be a classic case for a page in the [wiki.](../wiki) where we can link to.
Just someone would have to dare to create the first page 😏

sohamakl, 2020-01-05:
My English is not perfect and I don't know how to use a wiki on github yet.

But it seems to be a good start to write a little how-to on the topic "Hardware extension for time-controlled on/off switching of the Sunfounder 10.1 inch HDMI touch display".

If you agree, I would take it over.

Who can give me the necessary rights?

LukeSkywalker92, 2020-01-05:
you should be able to edit the wiki now.

gegu, 2020-01-23:
@sohamakl great wiki page 👍 

Closed by #97.

closed 2020-01-23 by gegu

sohamakl, 2020-01-25:
[image]

PR #94 (2020-01-23): Sunfounder screen config
closes #93