LukeSkywalker92/TeleFrame #105: Issue with actual develop branch and downloading images from Telegram

sohamakl, 2020-01-25:
I have problems with the current branch `develop`.  When I start TeleFrame via SSH with `tf.sh` and then send a picture I get the following error: 

[image]

Does anyone have any idea what going wrong?

gegu, 2020-01-25:
I can't reproduce the problem using your steps.

Is the image displayed in TeleFrame?

The error occurs during download in a node module. `substr` is not used in bot.js.

You can change line 150 in _bot.js_ to additionally log a stack trace:
```patch
@@ -145,11 +145,11 @@ var Bot = class {
                 chatName,
                 ctx.message.message_id
               );
             })
             .catch((err) => {
-              this.logger.error(err);
+              this.logger.error(err.stack);
             });
           }
         })
         .catch((err) => {
           this.logger.error('Download: ' + err.toString());
```

sohamakl, 2020-01-25:
No, the image is not displayed in TeleFrame. Adding `.stack` in line 150 does not change the error message. The images directory is empty.

By the way, any idea hot to scale images in github's wiki?

sohamakl, 2020-01-25:
I have found the solution.  I just forgot to switch off the privacy mode for the bot in Telegram `/setprivacy disable`.  Since I use a Telegram group. 

Maybe you can give an additional hint to this option in the error message.

gegu, 2020-01-25:
It would be nice to find out where the error occurs.

Could you please change line 155 as follows and reproduce the error
```patch
@@ -150,11 +150,11 @@ var Bot = class {
               this.logger.error(err);
             });
           }
         })
         .catch((err) => {
-          this.logger.error('Download: ' + err.toString());
+          this.logger.error('Download: fileId: ' + fileId + '\n' + err.stack);
           ctx.reply('Sorry: ' + err.toString());
         });
     });
 
     this.bot.catch((err) => {
```

sohamakl, 2020-01-25:
With the change I get the following message:

[image]

But now I'm sure to know the reason. After I configured the bot in Telegram, I assigned it directly to a group. I then sent the first photo out of the group.

This does not work until you have sent a picture directly to the bot. Afterwards it goes then also over the group.

An alternative could be to start the bot from within the group by command `/start`.

sohamakl, 2020-01-25:
Issue could be closed.

closed 2020-01-25 by gegu