LukeSkywalker92/TeleFrame #107: Isn't it time for a new master?

sohamakl, 2020-01-25:
Isn't it time for a new master? Several of the outstanding issues could be closed.

gegu, 2020-01-25:
Yeah, but I'm working on a new feature in branch [feature/addon-interface](https://github.com/LukeSkywalker92/TeleFrame/tree/feature/addon-interface). I'd rather wait until the branch is almost finished and has been transferred to develop.

sohamakl, 2020-01-25:
Can you say something more about it? Addon interface sounds good. Do you have a first addon in mind?

gegu, 2020-01-25:
Yes, for example a notification LED for new pictures.

I implement this so that, for example, hardware extensions can be implemented without having to change the core. In the end there should not be many limits to the creativity.

I will also provide some examples. Stay tuned, it's coming soon.

sohamakl, 2020-01-25:
This sounds very intersting. If you like you can close this issue, too.

sohamakl, 2020-01-25:
I have tested the current develop branch on Rasberry Pi 3 and 4 with Raspberry Pi 7'' DSI and sunfounder 10'' display with opto coupler hardware all with success.

Im running node js 13.7.0, npm 3.13.6 and electron 7.1.10.

Very nice, I'm so happy about this project :-)

closed 2020-02-13 by gegu