LukeSkywalker92/TeleFrame #109: TeleFrame addon interface

gegu, 2020-01-31:
First of all: Since some changes were necessary in this PR, I will not take it into the develop branch before @LukeSkywalker92  has given his consent. If it is not acceptable, I would create a fork.

With #108 I want to present you the idea of an addon interface.

The plan was to provide a simple way to extend the functionality of TeleFrame.

My idea was to create an easy to use interface.
When implementing it, I didn't want to affect the core too much and without additional dependencies.

Nevertheless, there should be as few restrictions as possible when implementing an add-on. In principle, an addon has access to almost all objects of the TeleFrame main process if you need them. See demo [TeleFrame-webRemote](https://github.com/gegu/TeleFrame-webRemote) (self-promotion :-))

You can read the details in the [README of the addon interface](https://github.com/LukeSkywalker92/TeleFrame/tree/feature/addon-interface/addons).

With the implementation it became also possible to use TeleFrame in a *frame-only* mode, without bot configuration  
- config: `{ "botToken": "bot-disabled" }`

This could be used to update images and `images.json` on the TeleFrame by implementing a sync addon. Simply copying the files and restarting the TeleFrame would also be possible, but is rather uncomfortable.

Let me know what you think. Suggestions for changes or additions are welcome.

Happy addon coding!

### Discussion: List of available addons

Since you will probably create some addons in the future, it would be IMO useful to have a list of available addons.

How do you think we could implement this in a meaningful way?

LukeSkywalker92, 2020-01-31:
For me this sounds great. Even if it breaks some older stuff.. We made so many changes that we can go for a new major release..
I hope to find some time to look deeper into your changes but you have my OK for it.
Thaks for the work :+1:

gegu, 2020-01-31:
Okay, thanks.

I also think it would be good to release soon.

Due to problems with the update of electron, an update from the current master is only really possible with manual intervention by deleting the node_modules/electron directory and setting the environment variable `npm_config_arch` .

I have revised the update instructions once again.

electron had a nasty problem with a memory leak until the current version 7.1.11. Now it seems to work.

Nothing should have broken, I hope.  The control without touchbar still works if one configures `touchBar=null`. 

Most changes to the existing files affected additional and modified event handlers for communication with the addon interface.

gegu, 2020-01-31:
> 
> 
> Even if it breaks some older stuff.. 

OK, I missed the botonly mode when testing. It is working again now and I have also cleaned up a bit.

sohamakl, 2020-01-31:
Very cool stuff. We could user a Wiki page to list available addons.

gegu, 2020-02-01:
I agree, the wiki is the best place for the list.

gegu, 2020-02-09:
The addon Interface is now finished and merged into develop.

@LukeSkywalker92  from my side, we would be ready to release.

LukeSkywalker92, 2020-02-10:
Sounds good. Thanks again for your work. Should we go for 3.0.0?

gegu, 2020-02-10:
I think we can do this. There have really been a lot of improvements in the last few weeks - both visible and under the hood.

gegu, 2020-02-11:
I' ve added a small change in the installscript so that git ignores changes of the filemode.

LukeSkywalker92, 2020-02-13:
Then go for it :smiley:

gegu, 2020-02-13:
It's done 😁

closed 2020-02-13 by gegu