LukeSkywalker92/TeleFrame #114: Increasing intervals

call-me-matt, 2020-03-01:
I had once a digital picture frame from canon and it had a feature which I liked: when interacting with the frame (startup, new pictures, touch etc.) the intervals between picture changes were short, and with time grew longer. This way the frame was less disturbing when in the background and more active when you were sitting in front of it. I think it would be a great feature for this implementation, too.

call-me-matt, 2020-03-02:
Here is a first implementation in form of an addon:
Any comments?
How should I submit my code?

```/**
 * introduces quicker image changes after interactions and slows down with time
 * @param  {AddonBase inherited} interface   object to register and send events
 */

// import interval from config.json
var fs = require('fs');
var configfile = fs.readFileSync("./config/config.json", "utf8");
var config = JSON.parse(configfile);
var maxInterval = (config.interval) ? config.interval:45000;
const additionalChanges = (config.imageCount) ? config.imageCount:30;
var counter = 1;

const functionFlexIntervals = (interface) => {

  //interface.registerListener('newImage', (sender, type) => counter = 1);

  interface.registerListener('images-loaded', () => counter = 1);

  interface.registerListener('imageDeleted', () => counter = 1);

  interface.registerListener('paused', (status) => counter = 1);

  interface.registerListener('muted', (status) => counter = 1);

  interface.registerListener('recordStopped', () => counter = 1);

  interface.registerListener('unstarImage', index => counter = 1);

  interface.registerListener('starImage', index => counter = 1);

  interface.registerListener('changedActiveImage', index => {
        // introduce an additional picture change:
        if (counter < additionalChanges) {
		var additionalChange = Math.round(maxInterval*counter/additionalChanges);
                interface.logger.info("additional change in: " + additionalChange);
		setTimeout(() => { interface.sendEvent('next'); }, additionalChange);
		counter += 1;
	};
  });
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {
  module.exports = functionFlexIntervals;
}```

gegu, 2020-03-03:
Nice idea, exactly for these requirements the addon interface was implemented.

A few hints:
You do not need to read the configuration file itself. The configuration is provided by the addon interface and all keys have a valid value. So you can access the values directly.

_For example:_
```js
//...

const functionFlexIntervals = (interface) => {
  var config;
  var counter = 1;

  interface.registerListener('teleFrame-ready', teleFrameObjects => {
    config = teleFrameObjects.config;
  });

//...

  interface.registerListener('changedActiveImage', index => {
    // introduce an additional picture change:
    if (counter < config.imageCount) {
      var additionalChange = Math.round(config.interval*counter/config.imageCount);
      interface.logger.info("additional change in: " + additionalChange);
      setTimeout(() => { interface.sendEvent('next'); }, additionalChange);
      counter += 1;
    };
  });
```

>
> How should I submit my code?

Please create a repository for the addon yourself and we will link it on the [Available TeleFrame addons - wiki page](https://github.com/LukeSkywalker92/TeleFrame/wiki/Available-TeleFrame-addons).

call-me-matt, 2020-03-03:
Cool!
Here you go:
https://github.com/call-me-matt/teleframe-flexintervals

gegu, 2020-03-04:
It' s done.

gegu, 2020-03-04:
Another little hint:
You can specify the events you want to listen to using an array if you want to do the same thing every time.
```js
interface.registerListener([
  'images-loaded', 'imageDeleted', 'paused', 'muted', 
  'recordStopped', 'unstarImage', 'starImage'
], index => counter = 1);
```

call-me-matt, 2020-03-04:
Thank you so much for your help.
I really like this addon-interface.

Thought about the description in the wiki, maybe "introduces quicker image changes which slow down with time" is a catchier description.

PS: You re-opened the issue?

gegu, 2020-03-05:
I'd enjoy it.

>
> Thought about the description in the wiki

You can edit the Wiki page yourself 😃 

>
> PS: You re-opened the issue?

Yes, because @LukeSkywalker92  has marked the extension as a enhancement and I also think it's a feature that is interesting for everyone. I am currently looking for an easy way to install addons that are interesting for everyone. There are some ideas already, but not yet fully realized.
So that the topic is not forgotten, I have reopened the issue.

call-me-matt, 2020-03-09:
is there a way to detect if a picture or video is shown?
(call-me-matt/teleframe-flexintervals#3)

gegu, 2020-03-09:
I suppose you've noticed the problem with video playback time. It almost drove me crazy yesterday, too.

You can use `interface.images[index].src` to check for `.mp4`

These changes should solve the problem
```diff
@@ -68,13 +68,16 @@ const functionFlexIntervals = (interface) => {
     }
     const additionalChange = Math.max(minChangeTime, Math.round(config.interval*counter/config.imageCount));
     nextChangeTime = new Date().getTime() + additionalChange;
     // introduce an additional picture change:
     if (!isPaused && counter < config.imageCount) {
-      setChangeTimer(additionalChange);
+      // ignore videos because the duration is unknown
+      if (!interface.images[index].src.match(/\.mp4$/i)) {
+        setChangeTimer(additionalChange);
+      }
       counter += 1;
-    };
+    }
   });
 };
```

call-me-matt, 2020-03-09:
Great, thank you

closed 2020-07-06 by LukeSkywalker92