LukeSkywalker92/TeleFrame #115: adminAction does not trigger

andrestricker, 2020-03-02:
I have set up my config file as follows:

whitelistAdmins:[**mychatidhere],
adminAction: {
    allowAdminAction: true,
    actions: [
     {name: "startvpn", command: "/home/pi/vpnc.sh", enable: true},
    ]
  }

but the action is not triggered. Tried to put 

/startvpn and
startvpn 

into the bot chat but no success. Triggers no logging. "Hi" returns the correct ChatID.

LukeSkywalker92, 2020-03-02:
Try `sh /home/pi/vpnc.sh` as command.

andrestricker, 2020-03-02:
Thanks for the quick reply. No this did not work. If I understand the code correctly, "/reboot" should be enabled by default and I should be able to trigger it as soon as my chat ID is in the whitelistAdmins array, right? It does not trigger as well.

andrestricker, 2020-03-02:
Removed the excess comma behind the action. Still no triggering

LuHKae, 2020-03-03:
If you have added an "actions"-Array in your config.json, all standard actions are overwritten.

So your config.json should be something like

…
"adminAction":{
	"actions":[
		{"name": "startvpn", "command": "/home/pi/vpnc.sh", "enable": true},
		{"name": "echo", "command": "echo Test", "enable": true},
		{"name": "reboot", "command": "sudo reboot", "enable": true},
		{"name": "shutdown", "command": "sudo shutdown -h now ", "enable": true},
		{"name": "restart", "command": "pm2 restart all", "enable": true}
	]
  }
…

Could you test this config.json and provide the TeleFrame-Log?

andrestricker, 2020-03-04:
I placed your snippet into my config.js. but no success. The only command that seems to work is "hi" and sending images to the frame.

This is the the Teleframe-Errorlog
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/
ALSA lib pcm_hw.c:1903:(_snd_pcm_hw_open) card is not defined
ALSA lib pcm_hw.c:1903:(_snd_pcm_hw_open) card is not defined

And this is the normal log:
--- 8.8.8.8 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 10.049/10.049/10.049/0.000 ms

> teleframe@2.0.1 start /home/pi/TeleFrame
> electron .

2020-03-04T19:51:13.622Z [Main] info: Main app started ...
2020-03-04T19:51:13.954Z [Main] info: Bot created!
2020-03-04T19:51:13.990Z [Main] info: Bot started!
2020-03-04T19:51:14.237Z [Main] info: Using bot with name *****_bot.
2020-03-04T19:51:15.397Z [Renderer] info: Renderer started ...

andrestricker, 2020-03-04:
This is my full config.js is - maybe I missed something there:

var config = { botToken: '****', 
  whitelistChats:[****], 
  whitelistAdmins:[****],
  playSoundOnRecieve: "sound1.mp3", 
  showVideos: false, 
  playVideoAudio: false, 
  imageFolder: "images", 
  fullscreen: true, 
  fadeTime: 1500, 
  interval: 10 * 1000, 
  imageCount: 100, newPhotoMessage: "Neues Foto von", 
  newVideoMessage: "Neues Video von", 
  showSender: true, 
  showCaption: true, 
  toggleMonitor: false, 
  turnOnHour: 8, 
  turnOffHour: 22, 
  keys: {
    next: "right",
    previous: "left",
    play: "l",
    pause: "k"
  },
  voiceReply: {
    key: "a",
    maxRecordTime: 60*1000,
    recordingMessageTitle: "Voice Message",
    recordingPreMessage: "Recording for",
    recordingPostMessage: "in progress...",
    recordingDone: "Voice message sent sucessfully!",
    recordingError: "Voice message has failed!"
  },
adminAction:{
  allowAdminAction: true,
  actions:
        [
      {name: "startvpn", "command": "sh /home/pi/vpnc.sh", "enable": true},
      {name: "echo", "command": "echo Test", "enable": true},
      {name: "reboot", "command": "sudo reboot", "enable": true},
      {name: "shutdown", "command": "sudo shutdown -h now ", "enable": true},
      {name: "restart", "command": "pm2 restart all", "enable": true}
    ]
  }
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {
  module.exports = config;
}

LuHKae, 2020-03-05:
Seems like you are using a wrong  Version. TeleFrame 3.x is using a json file for configuration and not a js file.

andrestricker, 2020-03-11:
Thanks, installed the latest version. Works like charme!

closed 2020-07-06 by LukeSkywalker92