LukeSkywalker92/TeleFrame #121: Fehler bei Installation via automatischem Script

BaseJam, 2020-03-13:
Umgebung:
Raspberry Pi 3 Model B Rev 1.2
Raspbian GNU/Linux 10 (buster)

Grundinstallation, danach "Automatic Installation (Raspberry Pi only!)"-Script ausgeführt.
Folgender Output:

Do you want to disable the screensaver (y/N)? y
Do you want to your mouse pointer do be autohided (y/N)? y
Do you want use pm2 for auto starting of your TeleFrame (y/N)? y
Do you want pm2 to wait for internet connection before auto starting your TeleFrame (y/N)? y
Please tell me your telegram bot token. Token:  12345678903
Updating packages ...
OK:1 http://archive.raspberrypi.org/debian buster InRelease
Holen:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15,0 kB]
OK:3 https://deb.nodesource.com/node_10.x buster InRelease
Es wurden 15,0 kB in 2 s geholt (8.631 B/s).
Paketlisten werden gelesen... Fertig
Installing helper tools ...
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
build-essential ist schon die neueste Version (12.6).
curl ist schon die neueste Version (7.64.0-4+deb10u1).
git ist schon die neueste Version (1:2.20.1-2+deb10u1).
libsox-fmt-all ist schon die neueste Version (14.4.2+git20190427-1).
sox ist schon die neueste Version (14.4.2+git20190427-1).
unclutter ist schon die neueste Version (8-21).
unzip ist schon die neueste Version (6.0-23+deb10u1).
wget ist schon die neueste Version (1.20.1-1.1).
x11-xserver-utils ist schon die neueste Version (7.7+8+rpi1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v10.15.2
Installed Node version: v10.19.0
No Node.js upgrade necessary.
Cloning TeleFrame ...
Klone nach 'TeleFrame' ...
remote: Enumerating objects: 85, done.
remote: Counting objects: 100% (85/85), done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 85 (delta 6), reused 46 (delta 0), pack-reused 0
Entpacke Objekte: 100% (85/85), Fertig.
Cloning TeleFrame Done!
Installing dependencies ...

> core-js@3.6.1 postinstall /root/TeleFrame/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

> electron@7.1.10 postinstall /root/TeleFrame/node_modules/electron
> node install.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module '/root/TeleFrame/node_modules/electron/install.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:623:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron@7.1.10 postinstall: `node install.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the electron@7.1.10 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-12T23_15_33_536Z-debug.log
Unable to install dependencies!

Issue#39 habe ich bereits versucht, neuste NPM ist installiert. Dennoch selber fehler.

BaseJam, 2020-03-13:
Lag wohl daran, dass ich die Installation mit dem Root-User durchgeführt habe. Hier wäre es super, wenn das Script das abfängt und einen Hinweis gibt.

gegu, 2020-03-13:
Mit dem nächsten Major-Update wird geprüft, ob das Installscript als root gestartet wurde und in diesem Fall mit einer Fehlermeldung beendet.

closed 2020-03-13 by gegu