LukeSkywalker92/TeleFrame #124: Touchbar isn't displayed

MaxEco, 2020-04-10:
Hi,
I have done a fresh install of Teleframe using the automatic installation on a Rapsberry Pi 4 with the latest version of Raspbian (2020-02-13 buster full). 
Once the installed was done and fully configured, I connected a touch screen to the Rapsberry in order to use the Touchbar and try if I was able to delete some pictures sent to the bot.

However when I touch the screen, the touchbar doesn't appear on it (I've added the "touchbar" parameters in config.json).
Am I missing something in the configuration ?

By the way the touchscreen I use for testing is a Luvox F190 and is not yet calibrated (I don't know if it can be an issue but I don't think so).

Thanks,

ForgottenTales, 2021-03-11:
I am having the same issue. It would be nice to get some help. But I have some more details.

The problem is not with TeleFrame itself. I also use a 7 inch touchscreen display. And it seems that somehow when booting up with Teleframe enabled it does not recognize the touch input. After unplugging the LCD and plugin it in again it works and also the touchbar appears. But witouht TeleFrame the touch input is always working after a system start. So somehow, the Teleframe autostart seems to break the touch input. But why?

JanSHUP, 2021-04-02:
Hi all, 

I think I have the same problem with a Waveshare 10.1 touchscreen.

The problem seems to be that the touchscreen is not interpreted as a touch device but as a HID mouse by Telegram / Linux. (In some threads it is said that this is standard for resistive touch screens in contrast to capacitive touchpanels which are interpreted as touch-events. But this seems to me to be a driver problem rather than a hardware limitation).

I replaced the touch events "touchend" in touchBar.js and renderer.js with "mousemove" and voila, the touchbar appears. 
Unfortunately, this is not the final solution: after all, mouse clicks are not interpreted and the touchbar flickers instead of appearing constantly for a certain time.

I think there are two solutions: 
1) Achieve a different interpretation of the touchpad input at the driver level.
2) make the teleframe-touchbar respond to mouse-move and mouse-click as an alternative to "touch-events".

unfortunately I lack the experience with .js frontend to implement it. If someone can help i would be very happy. The teleframe will be a present for my grandfather and the functions "next" and "delete" would be very practical.

best regards, Jan

JanSHUP, 2021-04-02:
Preliminary suggestion in case of this problem:
replace "touchend" with "click" in:
touchBar.js (2x)
renderer.js

The only disadvantage: The timer does not work, you have to click on the screen again to make the touchbar disappear. I can live with that.

This way, a Waveshark 10.1 touchscreen works fine.

gegu, 2021-04-02:
Here we go 😃 

I changed the event handling in touchBar.js to 'click' instead of 'touchend'. With this everything should actually work.

>
>The only disadvantage: The timer does not work, you have to click on the screen again to make the touchbar disappear. I can live with that.

The timer to hide the touchbar also worked as expected in my tests - not on a rasbperry Pi. I wouldn't see any reason why it shouldn't work either.

JanSHUP, 2021-04-03:
You are right, it works perfectly fine now.

My touchbar timer was mistakenly set so hight that I never waited until it disappeared.

Thanks for fixing!

MaxEco, 2021-04-05:
Hi everyone,

Thanks for the fix I'll give it a try but because it works for both of you with your configs, I think it will work for me as well. So I will close the issue.

Thanks again.

closed 2021-04-05 by MaxEco