LukeSkywalker92/TeleFrame #125: Mute button does't mute video sound

pumuqui, 2020-04-11:
Hello.
This days I set up my first TeleFrame and after some startup problems its working really great. I am just wondering if the Mute button should not turn off the video sound? I know that I can turn on or off the video sound in the TeleFrame configuration, but it would be better that the user could turn it on or off. Is this possible?
My hardware: Raspberry Pi 3 A+, Official 7” Touchscreen Display, WM8960 Audio HAT, Huawei E3531 3G USB stick.
Thanks for any help.

pumuqui, 2020-04-19:
Found out myself:
```
pi@raspberrypi:~/TeleFrame/js $ git diff renderer.js
diff --git a/js/renderer.js b/js/renderer.js
index c2d9f45..da4c005 100644
--- a/js/renderer.js
+++ b/js/renderer.js
@@ -646,7 +646,7 @@ function loadImage(isNext, fadeTime, goToLatest = false) {
   var $asset;
   if (image.src.split(".").pop() == "mp4") {
     $asset = $("<video/>");
-    $asset.prop('muted',  !config.playVideoAudio);
+    $asset.prop('muted',  !config.playVideoAudio || isMuted);
     $asset.prop('autoplay', true);
   } else {
     $asset = $("<img/>");
```

closed 2020-04-19 by pumuqui

thoschneider, 2020-12-20:
Great! Thanks a lot! This obvious functionality should be added to the main branch. Consider a pull request.