LukeSkywalker92/TeleFrame #131: Using old android phone as client

philffm, 2020-07-04:
Hi ! I was wondering if somebody already tried to hook up an old phone for that project? As my main Raspberry Pi (Buster Lite) is standing in the office but the picture frame will be in the kitchen and maybe somewhere else as well.
Any experiences with using a virtual screen via VNC or something in that direction? Or is it possible to access some url in the local network? 
Have some phones flying around and wanted to repurpose them 👍 

Thanks in advance ! Will also do some research on what would be the best approach.

gegu, 2020-07-05:
You can try [TeleFrame-webRemote](https://github.com/gegu/TeleFrame-webRemote)  Addon. This should cover your needs.

philffm, 2020-07-05:
Smooth! Still need to start a VNC session once to get the `DISPLAY= :0 nohup npm start &` command running. Maybe Xvfb (virtual screen) works as well but that workaround seems to work for now 🎉

[image]

closed 2020-07-06 by LukeSkywalker92