LukeSkywalker92/TeleFrame #134: Toggle Screenn On / Off over Telegram

tommy86geis, 2020-07-12:
Is ist possible to switch Display on and off over telegram?

kris711, 2020-07-13:
Hi yes this works. Please check 
````
screenConfig
toggleMonitor
turnOnHour
turnOffHour
````
in config\config.json

Trenar, 2020-07-13:
You can check what toggles your display (trace down the toggleMonitor key, kris711 mentioned above) and execute these commands as an admin action. This is what I did

tommy86geis, 2020-07-16:
ok, how can i check that out? can you give me your code? i don't know how the admin action is working.

Trenar, 2020-07-16:
providing my code would not help, since I use a very special display, that works a little different (Sunfounder 10.1")

Have a look at the files in `/config/screens/`. I guess, you're using a default HDMI screen (i.e. using `hdmi_default.js`. There you find the bash commands for turning on and off the screen's backlight (`cmdBacklightOff` and `cmdBacklightOn`). With these commands you can add two admin actions to your config file. See the readme for how to use admin actions or look at the examples in the `/js/defaultConfig.js`

kris711, 2020-07-17:
maybe its easier: Which hardware (Monitor) do you use?

tommy86geis, 2020-07-17:
I use a normal Monitor over HDMI. 
Only Switch Signal Off and on. So the Monitor switch off If there is no Signal

Trenar, 2020-07-17:
Good. In this case, the config lines I suggested above should work for you. To toggle your monitor, send "/turnDisplayOff" or "/turnDisplayOn" to your bot

tommy86geis, 2020-07-17:
It is working perfectly. Thanks a Lot

closed 2020-07-27 by LukeSkywalker92