LukeSkywalker92/TeleFrame #136: Shutdown Option

KateiRen, 2020-07-25:
# Shutdown Option

Wouldn't it be a good idea to add a _**Return to Desktop**_ option when clicking on the _**Shutdown**_ button?

While TeleFrame is already running, I am still struggling to get the usb mic and the I2S sound working properly. Jumping to the Desktop and trying the default controls for audio might have helped me faster then  tweeking around in Putty.

Thx for considering...

LukeSkywalker92, 2020-07-27:
Hi, sounds good...
As a workaround you can exit fullscreen with F11.

KateiRen, 2020-07-27:
Very true. Attached a keyboard and I could minimize the window.

### I have another topic with the restart/shutdown functions:
Today, I installed **TeleFrame on an ubuntu** tablet (had to tweek the path in ~/TeleFrame/tools/pm2_TeleFrame.json) and it runs very well. However, for some reason the **executeSystemCommand seem not to be working** (nothing happens when pressing these buttons and confirming the action).

I have no idea what might be missing here. There is a warning in the console though: 
`Electron Security Warning (insecure Content-Security-Policy) `
Not sure if this is the case on raspi as well or is specific to my manual installation (had to use sudo to install electron globally).

Any ideas here?
Thx!!

LukeSkywalker92, 2020-07-29:
Hi, the warning should not be a problem.

I think the permissions of the user running teleframe are causing your issue.
On a raspi, the user pi has way more permissions than a standard user on ubuntu (for example running sudo commands, without typing a password..).
Maybe you can try running TeleFrame as root (just for testing, you should not do this in general) and see if the commands are executed.

Sure, you are welcome to contribute to the "Building a teleframe" section =)

KateiRen, 2020-07-29:
I added a pipe to the executeSystemCommand to write an error log and found out exactly what you also guessed. Executing the sudo command asks for a password, which fails  obviously. Fixed that behaviour and TF on Xubuntu tablet works like a charm now.

Thanks a lot.

LukeSkywalker92, 2020-07-30:
I'm happy that you could solve your problem. Have fun with the TeleFrame =)

closed 2020-07-30 by LukeSkywalker92