LukeSkywalker92/TeleFrame #139: [feature request] friendly reminder

call-me-matt, 2020-07-30:
I would like the teleframe to remind me with a short telegram message if there have been no new photos for more than two weeks.

Probably this could be an addon simply resetting a timer on `newImage` listener - but a `sendTelegramMsg` input event would be needed.

gegu, 2020-07-30:
It should be possible to implement the required function in the addon.

You need a listener for the event `teleFrame-ready`. There you can save a reference to the telegram object of the bot for further use.

```js
const function sendImageReminder = (interface) => {
	const telegram = null;

	interface.registerListener('teleFrame-ready', teleFrameObjects => {
	  telegram = teleFrameObjects.bot.telegram;
	});

}
```

call-me-matt, 2020-07-30:
So, here we go!
https://github.com/call-me-matt/teleframe-imageReminder

Could you help me with l10n for the message?
And not sure if/how it works if multiple accounts or groups are in the whitelist...

gegu, 2020-07-31:
Unfortunately it is not possible to determine the language of a chatid. 
So you have to implement the handling yourself. For example you could use a standard message and optionally define the message to be sent in the config section of your addon for a specific chatid.

To handle the promise rejection implement a catch handler - see example code below.

Define messages for specific chatid's in`config.json`:
```json
{

  "addonInterface": {
    "addons": {
      "imageReminder": {
        "enabled": true,
          "reminderMessage": {
            "chatid-1":"Message for chatid-1",
            "chatid-2":"Message for chatid-2"
          }
        }
     }
  }
}
```

Example code (untested)
```js

for (i in config.whitelistChats) {
  let reminderMsg = "hey hey, please feed me with new photos 🤖";
  try {
    reminderMsg = interface.config.reminderMessage[config.whitelistChats[i]] || reminderMsg;
  } catch(_) {}

  // send message and catch errors
  telegram.sendMessage(config.whitelistChats[i], reminderMsg)
  .catch(error => {
    interface.logger.error(error);
  });
}
```

call-me-matt, 2020-07-31:
Ah, I see. Thank you for your instant reply! The catch for telegram was exactly what I was looking for.
Concerning the translation of the reminder, it's maybe a little too much, then. So I will introduce just one constant here.

call-me-matt, 2020-07-31:
I have added this to the [wiki](https://github.com/LukeSkywalker92/TeleFrame/wiki/Available-TeleFrame-addons). Took over the name you chose (imageReminder). Thank you @gegu !

closed 2020-07-31 by call-me-matt