LukeSkywalker92/TeleFrame #144: no botReply in group chats for some senders

junibart, 2020-11-02:
Hi, 

I just set up a TeleFrame for my grandmother, and created a chat group for the whole family...

The botReply message is activated in config.json, however some submissions by individuals do not receive a replymessage to the chat group. Only the three or four people I first included in the group trigger the message with their pics. 

The Teleframe log shows

`0|TeleFram | 2020-11-02T20:05:15.012Z [Main] info: Deleted file images/1603882250140.jpg`

for a picture with correct reply, and 

`0|TeleFrame  | 2020-11-02T20:10:28.239Z [Main] info: Deleted file images/1603892365865.jpg
0|TeleFrame  | 2020-11-02T20:10:28.245Z [Main] error: TypeError: Cannot read property 'substr' of undefined
0|TeleFrame  |     at botReply (/home/pi/TeleFrame/js/botReply.js:14:45)
0|TeleFrame  |     at /home/pi/TeleFrame/js/bot.js:131:19
`
for a picture without reply message. 

I want everyone in the group to receive a reply message. Is there something wrong with the code, how can it be fixed?

Thank you very much for developing TeleFrame, also in the name of my grandma and family ;-) it makes the current contact restriction a bit easier. 

Gunnar

call-me-matt, 2020-11-03:
Without looking at it in detail, have you tried to send "/start" from the non-working device to the bot?
I could imagine it to be a rights management issue, as bots are not allowed to contact random people.

LuHKae, 2020-11-03:
seems like one of your devices has no language set for the telegram app. Could you validate that?

gegu, 2020-11-03:
> 
> 
> seems like one of your devices has no language set for the telegram app. Could you validate that?

You're right 👍 

@junibart the problem was fixed, please update TeleFrame.

LukeSkywalker92, 2020-11-03:
nice job @gegu =) very fast... @junibart I'm happy to hear that the TeleFrame makes your family happy in these hard times :)

closed 2020-11-09 by LukeSkywalker92