LukeSkywalker92/TeleFrame #146: Add photos directly to images-Folder

Hazel202011, 2020-11-19:
Hi,

I would like to build a Teleframe and wondered if it is possible to directly put photos into the image-folder (via network share), so I could manually add media besides the Telegram bot.

Oh, and is it possible to play the sound of a video „on demand“? Default should be no sound but sometimes I may watch a video one time with sound.

Thanks in advise.
Hazel

LukeSkywalker92, 2020-11-27:
Hi,

the first is not implemented, but it should be possible to implement. So if you like... =)
Please create another issue as a feature request, maybe someone will pick it up...

With the Touch Bar you can turn sound on and off. Maybe this is already enough?

Cheers
Luke

gegu, 2020-11-27:
@Hazel202011 you could try the Addon [TeleFrame-webRemote](https://github.com/gegu/TeleFrame-webRemote) Here it is possible to upload pictures.

If you are familiar with javascript, It should also not be too difficult to develop an addon that monitors a directory and moves images into the TeleFrame images directory.

Hazel202011, 2020-12-07:
> @Hazel202011 you could try the Addon [TeleFrame-webRemote](https://github.com/gegu/TeleFrame-webRemote) Here it is possible to upload pictures.

Hi,
I just tried it out and it is perfect for my usecase. Thank you!

> With the Touch Bar you can turn sound on and off. Maybe this is already enough?

Yes, cool, this should be enough! Perfect!

I will open a feature request for the directory-feature, as I am not very familiar with programming :-D

Hazel

closed 2020-12-07 by Hazel202011