LukeSkywalker92/TeleFrame #148: Discussion: How to become a Teleframe-Admin

LuHKae, 2020-11-23:
Becourse of the editor free configuration it is in my opinion mandatory that at least one administrator is named for each TeleFrame installation. I would like to discuss with you how we can approach this.

Because it may be possible to manipulate the system with root privileges via the admin functions, it must be ensured that no stranger can become an admin.

My plan is to enable the frist sender of '/start' as admin during the setup of the teleframe.
In my eyes this should fit for all new installations.

More complicated is this topic for active installations. In my eyes there is no way to automatically query the admin.
I see three different ways on how this task could be acomplished:
1) After the update the first ID from whitelistChats becomes admin, which sends '/start'
2) After the update the ID from whitelistChats[0] is Admin
3) Don't let the bot start again if no admin has been maintained.

At the moment I think the best solution is a mix out of 1) and 2). 
If no admin has be configured so far, the bot will only start for a short time. Then the admin has this timeperiode to send his '/start' command, if he doesn't do that, the bot will terminate automatically.

What do you think about this?

LukeSkywalker92, 2020-11-27:
First of all...Nice Job 👍 

I like your suggestions and would vote for it if there are no arguments against it? @gegu ?

gegu, 2020-11-27:
Yes, I also think the idea of administration via chat is great.

However, it should definitely be possible to deactivate the feature and not necessarily pre-set it. Especially if we entered the security aspect.

For new installations the installer can ask if you want to use the feature and if the first chat id should be entered as admin in the whitelist. I would like that.

For existing installations, the update script may display a message during the update, telling you what to do to activate the feature. Here it would be conceivable that a script offers support for the subsequent activation/deactivation of the feature.
We could e.g. create a sempahore file in the config directory to see if the next /start should define the admin - or which idea comes up.  We could also implement a special command instead of /start to define the admin while the file exists.

Which I don't like when nothing works after an update, when there is no admin defined for a feature I might not want to use. I have installations running where I definitely don't want to.

I am available for further discussions on implementation. In about two weeks I have vacation and then some time to work on TeleFrame.

gegu, 2020-11-27:
>
>Which I don't like when nothing works after an update, when there is no admin defined for a feature I might not want to use. I have installations running where I definitely don't want to.

Sorry, what I wrote here was nonsense.  I have misinterpreted this 😟