LukeSkywalker92/TeleFrame #159: Telefram stratet nicht

seppbastel, 2021-01-09:
Hallo, ich bin über den Teleframe gestolpert und das ist genau das was ich suche für meine Schwiegereltern. 
Jetzt habe ich versucht TeleFrame zu installieren, die Installation ist durch gelaufen, aber es kommt kein TeleFrame zur Anzeige.
Hier Konfiguration usw.:
Raspberry 4 4GB
OS: Rasbian 
Node -v: v14.15.4
nodejs -v: v10.21.0
npm -v: 6.1.4.10
Config.json: 
"botToken": "13xxxxxxxx:Axxxxxxs",
  "whitelistChats": [],
 
Beim Start von npm Start erhalte ich folgende Fehlermeldung:
`
> teleframe@3.2.1 start /home/pi/TeleFrame
> electron .

App threw an error during load
SyntaxError: /home/pi/TeleFrame/config/config.json: Unexpected token : in JSON at position 12
    at JSON.parse (<anonymous>)
    at Object.Module._extensions..json (internal/modules/cjs/loader.js:906:27)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
    at require (internal/modules/cjs/helpers.js:68:18)
    at Object.<anonymous> (/home/pi/TeleFrame/js/configuration.js:92:15)
    at Module._compile (internal/modules/cjs/loader.js:880:30)
A JavaScript error occurred in the main process
Uncaught Exception:
SyntaxError: /home/pi/TeleFrame/config/config.json: Unexpected token : in JSON at position 12
    at JSON.parse (<anonymous>)
    at Object.Module._extensions..json (internal/modules/cjs/loader.js:906:27)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
    at require (internal/modules/cjs/helpers.js:68:18)
    at Object.<anonymous> (/home/pi/TeleFrame/js/configuration.js:92:15)
    at Module._compile (internal/modules/cjs/loader.js:880:30)
`
Browseraufruf funktioniert.
https://api.telegram.org/bot113xxxxxxxx:Axxxxxxs/sendMessage?chat_id=1xxxxxxxxx6&text=raspi

Wo liegt mein Denkfehler oder was habe ich falsch gemacht.
Vielen Dank für eure Hilfe

LukeSkywalker92, 2021-01-10:
Hi,

irgendwas scheint mit deiner config.json nicht zu passen.

kannst du mal den Inhalt der Datei `/home/pi/TeleFrame/config/config.json` zeigen? Natürlich nicht mit dem echten token...

seppbastel, 2021-01-10:
"botToken": "1xxxxxxx:Axxxxxxxxxxxxxxxxxxs",
  "whitelistChats": [],
  "whitelistAdmins": [],
  "screenConfig": "./config/screens/hdmi_default.js",
  "playSoundOnRecieve": "sound1.mp3",
  "showVideos": true,
  "playVideoAudio": false,
  "imageFolder": "images",
  "fullscreen": true,
  "fadeTime": 1500,
  "interval": 10000,
  "imageCount": 30,
  "autoDeleteImages": true,
  "showSender": true,
  "showCaption": true,
  "senderAndCaptionDuration": 50,
  "useFullscreenForCaptionAndSender": true,
  "toggleMonitor": false,
  "turnOnHour": 9,
  "turnOffHour": 22,
  "switchLedsOff": false,
  "touchBar": {
    "height": "75px",
    "autoHideTimeout": 30000,
    "elements": [
      "previousImage",
      "nextImage",
      "playPause",
      "record",
      "starImage",
      "deleteImage",
      "mute",
      "shutdown",
      "reboot"
    ]
  },
  "keys": {
    "next": "right",
    "previous": "left",
    "play": "l",
    "pause": "k"
  },
  "voiceReply": {
    "key": "a",
    "maxRecordTime": 60000
  },
  "adminAction": {
    "allowAdminAction": true,
          "actions": [
      {"name": "echo", "command": "echo test", "enable": true},
      {"name": "reboot", "command": "sudo reboot", "enable": true},
      {"name": "shutdown", "command": "sudo shutdown -h now ", "enable": true},
      {"name": "restart", "command": "pm2 restart all", "enable": true},
      {"name": "startOpenvpn", "command": "systemctl openvpn start", "enable": true},
      {"name": "stopOpenvpn", "command": "systemctl openvpn stop", "enable": true}
    ]
  }
}

LukeSkywalker92, 2021-01-10:
ok, wenn das genau der Inhalt der Datei ist, dann fehlt ein `{` am Anfang

seppbastel, 2021-01-10:
Vielen Dank, es funktioniert jetzt. Hatte die config.json vorher schon mal überprüft, aber die Klammer irgenwie übersehen.
Gibt es eine Möglichkeit die Anwendung zu unterbrechen und zum normalen Desktop zu gelangen? Möchte am System noch etwas einstellen. wenn ich über F1 LXTerminal öffne funktionieren ja bestimmte Tasten nicht.

seppbastel, 2021-01-10:
Kennst du das Problem, wenn ich ein bestehendes Video vom IPhone senden will, bekomme ich die Meldung "Dieses Dokument hat ein unbekanntes Dateiformat". Mache ich sozusagen ein Livevideo funktioniert das.