LukeSkywalker92/TeleFrame #175: stuck at splash screen "TeleFrame"

BongHole, 2021-11-06:
Hi,

First of all a big thank you to Luke!!
Today I created my 4th "TeleFrame"

but this one gives me some trouble… ☹

It actually works but it is not showing the pictures, there is just a black screen with TeleFrame shown!
The Bot gets pictures stores them even in the folder “image” but the slideshow doesn’t start

My config looks like

{
  "botToken": "teletoken",
 "whitelistChats": [69420],
  "whitelistAdmins": [69420],
"fullscreen": true,
  "fadeTime": 1500,
  "interval": 10000,
  "imageCount": 30,
  "autoDeleteImages": true,
  "showSender": true,
  "showCaption": true,
 "adminAction": {
    "allowAdminAction": true,
	  "actions": [
      {"name": "echo", "command": "echo test", "enable": true},
      {"name": "reboot", "command": "sudo reboot", "enable": true},
      {"name": "shutdown", "command": "sudo shutdown -h now ", "enable": true},
      {"name": "restart", "command": "pm2 restart all", "enable": true},
      {"name": "startOpenvpn", "command": "systemctl openvpn start", "enable": true},
      {"name": "stopOpenvpn", "command": "systemctl openvpn stop", "enable": true}
    ]
  }

}

Any idear what is wrong?

Thanks

BongHole

LukeSkywalker92, 2021-11-07:
Hey,

it's hard to tell what's going wrong without seeing the logs...
Can you please share them with us?

And please paste it as code for better readability =) (https://www.markdownguide.org/extended-syntax/#fenced-code-blocks)

Cheers
Luke

spychi, 2021-11-07:
Same bug on my setup.

Any idea?

BongHole, 2021-11-07:
Hi Luke thanks for your reply !

I hope that are the logs you are looking for :)

(node:1964) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
/home/pi/TeleFrame/node_modules/electron/dist/electron exited with signal SIGINT
npm WARN npm npm does not support Node.js v10.24.1
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm You can find the latest version at https://nodejs.org/
(node:2217) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
npm WARN npm npm does not support Node.js v10.24.1
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm You can find the latest version at https://nodejs.org/
(node:2385) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
npm WARN npm npm does not support Node.js v10.24.1
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm You can find the latest version at https://nodejs.org/
(node:2543) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
/home/pi/TeleFrame/node_modules/electron/dist/electron exited with signal SIGINT
npm WARN npm npm does not support Node.js v10.24.1
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm You can find the latest version at https://nodejs.org/
(node:2681) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
Failed to process updates. TelegramError: 409: Conflict: terminated by other getUpdates request; make sure that only one bot instance is running
    at /home/pi/TeleFrame/node_modules/telegraf/core/network/client.js:281:17
    at processTicksAndRejections (internal/process/task_queues.js:97:5) {
  code: 409,
  response: {
    ok: false,
    error_code: 409,
    description: 'Conflict: terminated by other getUpdates request; make sure that only one bot instance is running'
  },
  description: 'Conflict: terminated by other getUpdates request; make sure that only one bot instance is running',
  parameters: {},
  on: { method: 'getUpdates?offset=0&limit=100&timeout=30', payload: {} }
}

spychi, 2021-11-07:
Where can I find the log file?

BongHole, 2021-11-07:
this log file I found 
/home/pi/.pm2/logs/TeleFrame-error.log

I tried to update with

sudo apt-get update
sudo apt-get upgrade

but that also didnt help because everything was up to date!

I am  not realy used to linux, so it is more try and error :)

spychi, 2021-11-07:
Maybe i found a hint in JS Console 
renderer.js:9 Uncaught TypeError: Cannot read property 'getGlobal' of undefined
    at renderer.js:9

// Imports
const {remote, ipcRenderer, webFrame} = require("electron");
const $ = require("jquery");
window.jQuery = $;
const Swal = require("sweetalert2");
const randomColor = require("randomcolor");
const chroma = require("chroma-js");
const velocity = require("velocity-animate");
const logger = remote.getGlobal("rendererLogger"); // this is Line 9 
const config = remote.getGlobal("config");

spychi, 2021-11-07:
The problem come with the update to electron v11.
With version 9.4.0 still works.
You can change the version in the package.json and restart with npm install && npm start

BongHole, 2021-11-07:
nice!

works for me! 

thank you!

LukeSkywalker92, 2021-11-08:
Hey guys.

I fixed the problem with #176.
Now you can use the latest electron version again.

Version 9.4.0 has some security issues, so it is wise to upgrade.

Cheers
Luke

commit b977fae2 (2021-11-08): Merge pull request #176 from LukeSkywalker92/develop

Fix #175
closed 2021-11-08 by LukeSkywalker92