LukeSkywalker92/TeleFrame #177: Reboot fails

AAAbrAkAdAbrAAA, 2021-12-22:
First of all thanks for this Open Source Project.
It is awesome and I hope to make my grandma happy this Christmas with a frame :)

But, when installing TeleFrame and booting it first, everything works fine. 
Pictures are received and displayed.
So far, so good.

But when rebooting, the raspberry loops into infinity and input does not work.

I already disabled and deinstalled pm2, but it is still an issue. 
The raspberry tries to boot TeleFrame right from the start.

The TeleFrame Logo shows up first when rebooting. 
Then the screen goes off again.
Then the screen is on without any text or output.

LukeSkywalker92, 2021-12-23:
Hey,
I'm happy that you like the project!
I see you are in a little hurry =)

First of all: which pm2 script are you using? Did you choose the one that is waiting for internet connection?
Second: can you share the pm2 logs with us?

AAAbrAkAdAbrAAA, 2021-12-23:
Yes, this is definitely a last minute present :-D

I tried both versions, with and without waiting, both seem to stuck.

How can I reach the logs, when the pi is totally freezed?
Can I interrupt the pm2 startup?

LukeSkywalker92, 2021-12-23:
Do you have ssh access to the pi? Or is this also frozen?
If you have access you can use `pm2 logs 0` to get the logs...

AAAbrAkAdAbrAAA, 2021-12-23:
let me try :)

AAAbrAkAdAbrAAA, 2021-12-23:
Here are the log with the frame on first boot up, working fine:

pi@raspberrypi:~ $ pm2 logs 0
[TAILING] Tailing last 15 lines for [0] process (change the value with --lines option)
/home/pi/.pm2/logs/TeleFrame-error.log last 15 lines:
0|TeleFram | (node:9063) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information

/home/pi/.pm2/logs/TeleFrame-out.log last 15 lines:
0|TeleFram | 
0|TeleFram | > teleframe@3.2.3 start
0|TeleFram | > electron .
0|TeleFram | 
0|TeleFram | 2021-12-23T12:23:10.458Z [Main] info: Configuring for: Standard HDMI screen
0|TeleFram | 2021-12-23T12:23:10.476Z [Main] info: Main app started ...
0|TeleFram | 2021-12-23T12:23:13.218Z [Main] info: [AddonInterface] Load addons...
0|TeleFram | 2021-12-23T12:23:13.227Z [Main] info: [AddonInterface] Initialize listeners...
0|TeleFram | 2021-12-23T12:23:13.234Z [Main] info: [AddonInterface] Addons loaded
0|TeleFram | 2021-12-23T12:23:13.262Z [Main] info: Add Admin-Actions
0|TeleFram | 2021-12-23T12:23:13.269Z [Main] info: Bot created!
0|TeleFram | 2021-12-23T12:23:13.380Z [Main] info: Bot started!
0|TeleFram | 2021-12-23T12:23:14.089Z [Main] info: Using bot with name BilderrahmenOma_bot.
0|TeleFram | 2021-12-23T12:23:18.893Z [Renderer] info: Renderer started ...

LukeSkywalker92, 2021-12-23:
Ok, and then you do a reboot and it stops working?
It looks like the logs are from some minutes ago, so ist it working right now?

AAAbrAkAdAbrAAA, 2021-12-23:
Now I did the reboot and the pi is frozen.

I can ping the pi, but can not connect through SSH

AAAbrAkAdAbrAAA, 2021-12-23:
What is also strange to me:
I tried it also with pm2 disabled, but the TeleFrame starts as well here at start up. 
Is the autostart triggered anywhere else then?

I would like configure the frame the way, that it does not start from boot up.

LukeSkywalker92, 2021-12-23:
how did you disable it?
`pm2 stop 0` + `pm2 save`?

LukeSkywalker92, 2021-12-23:
if you don't want to run teleframe at boot, you don't need pm2...
then you can just run it manually

AAAbrAkAdAbrAAA, 2021-12-23:
I mean when setting up the frame. 
within one setup I said to pm2: No
But when I rebooted the frame, the application startet at startup.

LukeSkywalker92, 2021-12-23:
did you just run the install script again?

AAAbrAkAdAbrAAA, 2021-12-23:
yes, sevaral times.

LukeSkywalker92, 2021-12-23:
Ok, this might be the problem. I am honestly not sure what happens when you run it multiple times. My advice would be to flash a new raspberry pi os and run the script only once with the desired settings...

AAAbrAkAdAbrAAA, 2021-12-23:
okay, sorry. I mean totally from new, with new OS.

AAAbrAkAdAbrAAA, 2021-12-23:
It must be outside of the TeleFrame. 
No I deleted TeleFrame from the pm2 List.
And rebooted. But the raspberry runs into a freeze again. 

Is the installation of the teleframe changing anything else anywhere for the bootup?

LukeSkywalker92, 2021-12-23:
Not really. What happens when you reboot a new OS?

LukeSkywalker92, 2021-12-23:
But you could also try the manual installation...

Chrissib1743, 2022-03-17:
@AAAbrAkAdAbrAAA 
Have you ever resolved this issue?
I‘m having the same problem.

JanSHUP, 2022-04-01:
Hey everybody. 
I found the same issue with a Raspberry Pi 4b 4Gb with Raspberry Pi OS.
I installed with "install_raspberry.sh" via SSH and also got some NPM and NodeJs warnings. (See below).
Pi is not booting correctly anymore and running into black screen after ~5 seconds. 
SSH connection fails -> "login as:"-request and password-request are shown but after sending the password it is not responding anymore.

If anybody has an idea what went wrong or if I anyhow can get the PI back to life or avoid this behaivour at next installation I would be happy. Thanks in advance.

PS: Raspberry Pi OS was installed yesterday and kodi was compiled manually to run parallel to teleframe. Nothing else was changed systemwise.

Installation log:   (unfortunately it is german)

bash -c "$(curl -sL https://raw.githubusercontent.com/LukeSkywalker92/TeleFrame/master/tools/install_raspberry.sh)"

_________ _______  _        _______  _______  _______  _______  _______  _______
\__   __/(  ____ \( \      (  ____ \(  ____ \(  ____ )(  ___  )(       )(  ____ \
   ) (   | (    \/| (      | (    \/| (    \/| (    )|| (   ) || () () || (    \/
   | |   | (__    | |      | (__    | (__    | (____)|| (___) || || || || (__
   | |   |  __)   | |      |  __)   |  __)   |     __)|  ___  || |(_)| ||  __)
   | |   | (      | |      | (      | (      | (\ (   | (   ) || |   | || (  
   | |   | (____/\| (____/\| (____/\| )      | ) \ \__| )   ( || )   ( || (____/\
   )_(   (_______/(_______/(_______/|/       |/   \__/|/     \||/     \|(_______/

Do you want to disable the screensaver (y/N)? y
Do you want your mouse pointer to be autohided (y/N)? y
Do you want to use pm2 for auto starting of your TeleFrame (y/N)? y
Do you want pm2 to wait for internet connection before auto starting your TeleFrame (y/N)? n
Please tell me your telegram bot token. Token:  XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Updating packages ...
Holen:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15,0 kB]
OK:2 http://archive.raspberrypi.org/debian bullseye InRelease
Es wurden 15,0 kB in 1 s geholt (23,4 kB/s).
Paketlisten werden gelesen… Fertig
Installing helper tools ...
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
build-essential ist schon die neueste Version (12.9).
curl ist schon die neueste Version (7.74.0-1.3+deb11u1).
git ist schon die neueste Version (1:2.30.2-1).
unzip ist schon die neueste Version (6.0-26).
wget ist schon die neueste Version (1.21-1+deb11u1).
wget wurde als manuell installiert festgelegt.
x11-xserver-utils ist schon die neueste Version (7.7+8).
x11-xserver-utils wurde als manuell installiert festgelegt.
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  libfuse2
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
Die folgenden zusätzlichen Pakete werden installiert:
  libao-common libao4 libopencore-amrnb0 libopencore-amrwb0 libsox-fmt-alsa
  libsox-fmt-ao libsox-fmt-base libsox-fmt-mp3 libsox-fmt-oss
  libsox-fmt-pulse libsox3 unclutter-startup
Vorgeschlagene Pakete:
  libaudio2 libsndio6.1
Die folgenden NEUEN Pakete werden installiert:
  libao-common libao4 libopencore-amrnb0 libopencore-amrwb0 libsox-fmt-all
  libsox-fmt-alsa libsox-fmt-ao libsox-fmt-base libsox-fmt-mp3
  libsox-fmt-oss libsox-fmt-pulse libsox3 sox unclutter unclutter-startup
0 aktualisiert, 15 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 936 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 1.947 kB Plattenplatz zusätzlich benutzt.
Holen:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libao-common all 1.2.2+20180113-1.1 [11,7 kB]
Holen:2 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libao4 armhf 1.2.2+20180113-1.1 [33,6 kB]
Holen:7 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libsox-fmt-ao armhf 14.4.2+git20190427-2 [48,3 kB]
Holen:8 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libsox-fmt-base armhf 14.4.2+git20190427-2 [69,0 kB]
Holen:3 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian bullseye/main armhf libopencore-amrnb0 armhf 0.1.5-1 [85,3 kB]
Holen:11 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libsox-fmt-pulse armhf 14.4.2+git20190427-2 [47,9 kB]
Holen:13 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf sox armhf 14.4.2+git20190427-2 [139 kB]
Holen:4 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian bullseye/main armhf libopencore-amrwb0 armhf 0.1.5-1 [45,9 kB]
Holen:5 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libsox3 armhf 14.4.2+git20190427-2 [227 kB]
[... 196 lines cut ...]
remote: Counting objects: 100% (89/89), done.
remote: Compressing objects: 100% (81/81), done.
remote: Total 89 (delta 9), reused 44 (delta 1), pack-reused 0
Empfange Objekte: 100% (89/89), 11.54 MiB | 8.60 MiB/s, fertig.
Löse Unterschiede auf: 100% (9/9), fertig.
Cloning TeleFrame Done!
Installing dependencies ...
bash: Zeile 131: npm: Kommando nicht gefunden.
Unable to install dependencies!

###############
########### SCRIPT Stopped here and due to "NPM Comand not found" I tried to install nodejs and NPM again.
###############

pi@raspberrypi:~ $ sudo apt-get install -y nodejs
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
nodejs ist schon die neueste Version (12.22.5~dfsg-2~11u1).
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  libfuse2
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

pi@raspberrypi:~ $ reboot

JanSHUP, 2022-04-01:
reading the log again I guess I should have tried installing the npm package manually? 
at least the log suggests to install npm package:
"Vorgeschlagene Pakete:
  npm" 
and doesnt install automatically - at least that is what I read there.
Unfortunately I need to install Pi Os again to check if this is the real issue.

JanSHUP, 2022-04-02:
Hello together, 

I made a fresh PiOs installation and used the install.sh script again. got the same result as yesterday. Therefore I tried to install npm manually and try again. Unsuccessfull - see below.

~~~
sudo apt install npm
nano teleframe2.sh
~~~

I copied the install-script from line 131 where the error appeared into teleframe2.sh and finished installation

~~~
chmod +x teleframe2.sh
./teleframe2.sh
   
-> 
Installing dependencies ...

> teleframe@3.2.3 install
> node ./tools/install.js

added 197 packages, and audited 198 packages in 34s

6 vulnerabilities (1 low, 3 moderate, 2 high)

To address issues that do not require attention, run:
  npm audit fix

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.
Dependencies installation Done!
Installing electron globally ...

added 86 packages, and audited 87 packages in 29s

5 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
Electron installation Done!
Creating image directory ...
mkdir: das Verzeichnis ▒images▒ kann nicht angelegt werden: Die Datei existiert bereits
Check plymouth installation ...
./teleframe2.sh: Zeile 31: command_exists: Kommando nicht gefunden.
plymouth is not installed.

We're ready! Run DISPLAY=:0 npm start from the ~/TeleFrame directory to start your TeleFrame.

~~~
Now I tried to run teleframe: 

~~~
pi@raspberrypi:~ $ cd TeleFrame/
pi@raspberrypi:~/TeleFrame $  DISPLAY=:0 npm start

> teleframe@3.2.3 start
> electron .

2022-04-02T09:16:55.487Z [Main] info: Configuring for: Standard HDMI screen
2022-04-02T09:16:55.492Z [Main] info: Main app started ...
2022-04-02T09:16:55.909Z [Main] info: [AddonInterface] Load addons...
2022-04-02T09:16:55.910Z [Main] info: [AddonInterface] Initialize listeners...
2022-04-02T09:16:55.911Z [Main] info: [AddonInterface] Addons loaded
(node:5928) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
[... 303 lines cut ...]
21 timing command:run-script Completed in 5171ms
22 timing command:start Completed in 5180ms
23 verbose stack Error: command failed
23 verbose stack     at ChildProcess.<anonymous> (/usr/share/nodejs/@npmcli/promise-spawn/index.js:64:27)
23 verbose stack     at ChildProcess.emit (events.js:314:20)
23 verbose stack     at maybeClose (internal/child_process.js:1022:16)
23 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
24 verbose pkgid teleframe@3.2.3
25 verbose cwd /home/pi/TeleFrame
26 verbose Linux 5.10.92-v7l+
27 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
28 verbose node v12.22.5
29 verbose npm  v7.5.2
30 error code 1
31 error path /home/pi/TeleFrame
32 error command failed
33 error command sh -c electron .
34 verbose exit 1
~~~
I rebooted the PI with 
~~~
sudo reboot
~~~
and got again into the stucked mode I was yesterday. After SSH authentification SSH stops working (cant write or send).
After reboot I see the black Teleframe logo for a second and than it hangs again.

JanSHUP, 2022-04-02:
**Boot-Lock solved, NPM error still persistent**

Wondering that it still stuck beside the service beeing disabled I had a closer look at the installation script and made a change in /boot/config.txt having the SD card inserted into a Windows machine:

I commented the added dtoverlay
~~~
#dtoverlay=vc4-fkms-v3d
~~~
now it is booting again.

A first google search revealed, that it could be the combination of the newest Raspberry Pi OS with the display (mine is Waveshare 10.1 Touchscreen) and dtoverlay=vc4-fkms-v3d. 

Nevertheless, teleframe is still not starting. Showing the white starting screen shortly but crashing a second later after using 
~~~
npm start
~~~
In terminal again I get:
~~~
 DISPLAY=:0 npm start

> teleframe@3.2.3 start
> electron .

2022-04-02T13:40:12.868Z [Main] info: Configuring for: Standard HDMI screen
2022-04-02T13:40:12.872Z [Main] info: Main app started ...
2022-04-02T13:40:13.170Z [Main] info: [AddonInterface] Load addons...
2022-04-02T13:40:13.171Z [Main] info: [AddonInterface] Initialize listeners...
2022-04-02T13:40:13.174Z [Main] info: [AddonInterface] Addons loaded
2022-04-02T13:40:13.216Z [Main] info: Add Admin-Actions
2022-04-02T13:40:13.217Z [Main] info: Bot created!
2022-04-02T13:40:13.249Z [Main] info: Bot started!
(node:1578) electron: The default of contextIsolation is deprecated and will be                                                                              changing from false to true in a future release of Electron.  See https://github                                                                             .com/electron/electron/issues/23506 for more information
2022-04-02T13:40:13.491Z [Main] info: Using bot with name EJ_Fotorahmen_bot.
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-                                                                             bpf failure in syscall 0403
[1578:0402/144015.759339:FATAL:gpu_data_manager_impl_private.cc(445)] GPU proces                                                                             s isn't usable. Goodbye.
/home/pi/TeleFrame/node_modules/electron/dist/electron exited with signal SIGTRA                                                                             P
npm ERR! code 1
npm ERR! path /home/pi/TeleFrame
npm ERR! command failed
npm ERR! command sh -c electron .

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-04-02T13_40_15_881Z-debug.log
~~~

I don't know if this error is related to the boot problem and the overlay but I guess it is not.