LukeSkywalker92/TeleFrame #183: Automatic install & run issues with Raspberry Pi Os Bullseye

JanSHUP, 2022-04-03:
Hey together, 

the last days I tried to get TeleFrame running on Bullseye on a RPi 4b 4Gb but find some differences to Buster:

1) the automatic added "dtoverlay=vc4-fkms-v3d" is not used by Bullseye anymore. In my (and some others cases) the overlay results in a frozen boot state.

2) npm is not installed automatically with nodejs by the automatic installation script (see below)

3) I get some further problems installing / starting the teleframe (see below) e.g.:
- raspberrypi kernel: Unhandled prefetch abort: breakpoint debug exception (0x222)
- ../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
- [8096:0403/224531.695448:FATAL:gpu_data_manager_impl_private.cc(445)] GPU process isn't usable. Goodbye.
- /home/pi/TeleFrame/node_modules/electron/dist/electron exited with signal SIGTRAP

I couldn't get it running yet! I changed back to Buster for a comparison and everything ran as expected. 

Details:
Installation on a fresh Raspberry Pi OS Bullseye (official image from 2022.01.28):

~~~
pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/LukeSky                                                                             walker92/TeleFrame/master/tools/install_raspberry.sh)"

_________ _______  _        _______  _______  _______  _______  _______  _______                                                                             
\__   __/(  ____ \( \      (  ____ \(  ____ \(  ____ )(  ___  )(       )(  ____                                                                              \
   ) (   | (    \/| (      | (    \/| (    \/| (    )|| (   ) || () () || (    \                                                                             /
   | |   | (__    | |      | (__    | (__    | (____)|| (___) || || || || (__                                                                                
   | |   |  __)   | |      |  __)   |  __)   |     __)|  ___  || |(_)| ||  __)                                                                               
   | |   | (      | |      | (      | (      | (\ (   | (   ) || |   | || (                                                                                  
   | |   | (____/\| (____/\| (____/\| )      | ) \ \__| )   ( || )   ( || (____/                                                                             \
   )_(   (_______/(_______/(_______/|/       |/   \__/|/     \||/     \|(_______                                                                             /

Do you want to disable the screensaver (y/N)? y
Do you want your mouse pointer to be autohided (y/N)? y
Do you want to use pm2 for auto starting of your TeleFrame (y/N)? n
Please tell me your telegram bot token. Token:  XXXXXXXXXXXXXXXXXXXXXXXXXXX
Updating packages ...
OK:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
OK:2 http://archive.raspberrypi.org/debian bullseye InRelease
Paketlisten werden gelesen… Fertig
Installing helper tools ...
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
build-essential ist schon die neueste Version (12.9).
curl ist schon die neueste Version (7.74.0-1.3+deb11u1).
git ist schon die neueste Version (1:2.30.2-1).
unzip ist schon die neueste Version (6.0-26).
wget ist schon die neueste Version (1.21-1+deb11u1).
wget wurde als manuell installiert festgelegt.
x11-xserver-utils ist schon die neueste Version (7.7+8).
x11-xserver-utils wurde als manuell installiert festgelegt.
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  libfuse2
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
Die folgenden zusätzlichen Pakete werden installiert:
  libao-common libao4 libopencore-amrnb0 libopencore-amrwb0 libsox-fmt-alsa
  libsox-fmt-ao libsox-fmt-base libsox-fmt-mp3 libsox-fmt-oss libsox-fmt-pulse
  libsox3 unclutter-startup
Vorgeschlagene Pakete:
  libaudio2 libsndio6.1
Die folgenden NEUEN Pakete werden installiert:
  libao-common libao4 libopencore-amrnb0 libopencore-amrwb0 libsox-fmt-all
  libsox-fmt-alsa libsox-fmt-ao libsox-fmt-base libsox-fmt-mp3 libsox-fmt-oss
  libsox-fmt-pulse libsox3 sox unclutter unclutter-startup
0 aktualisiert, 15 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 936 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 1.947 kB Plattenplatz zusätzlich benutzt.
Holen:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libao-commo                                                                             n all 1.2.2+20180113-1.1 [11,7 kB]
Holen:2 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libao4 armh                                                                             f 1.2.2+20180113-1.1 [33,6 kB]
Holen:4 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian bullseye/main armh                                                                             f libopencore-amrwb0 armhf 0.1.5-1 [45,9 kB]
[... 563 lines cut ...]
26 timing npm:load:display Completed in 22ms
27 verbose logfile logs-max:10 dir:/home/pi/.npm/_logs
28 verbose logfile /home/pi/.npm/_logs/2022-04-03T20_45_48_273Z-debug-0.log
29 timing npm:load:logFile Completed in 17ms
30 timing npm:load:timers Completed in 0ms
31 timing npm:load:configScope Completed in 1ms
32 timing npm:load Completed in 85ms
33 silly logfile done cleaning log files
34 http fetch GET 200 https://registry.npmjs.org/npm 745ms (cache miss)
35 http fetch GET 200 https://registry.npmjs.org/npm 852ms (cache miss)
36 timing command:run-script Completed in 4742ms
37 timing command:start Completed in 4752ms
38 verbose exit 1
39 timing npm Completed in 4866ms
~~~

in journalctl -xe i find:

~~~
Apr 03 22:56:04 raspberrypi kernel: Unhandled prefetch abort: breakpoint debug exception (0x222) at 0x00000000
~~~

Do you have any ideas what could be wrong or how I could continue debugging?

Thanks

reiseradeln, 2022-06-20:
Hi @JanSHUP, 

I just tried to install TeleFrame on a new Raspberry Pi. In the end it worked. 

1. Update the Pi
`sudo apt-get update`
`sudo apt-get upgrade`

2. Install Node package manager - npm
`sudo apt-get install npm`

3. Install latest node version
`sudo npm install -g n@latest`

4. Install latest v8 version
`sudo npm install -g v8@latest`

5. Install TeleFrame
`bash -c "$(curl -sL https://raw.githubusercontent.com/LukeSkywalker92/TeleFrame/master/tools/install_raspberry.sh)"`

6. Prevent boot freeze by comment out the last line in config.txt.
`sudo nano /boot/config.txt`
`#dtoverlay=vc4-fkms-v3d`

7. Prevent stuck at TeleFrame startscreen (Thanks to @celphicious)
`cd TeleFrame/`
`nano main.js`
-> Add the bold marked line after line 42
```
	// Create the browser window.
	win = new BrowserWindow({
	width: 1024,
	height: 600,
	webPreferences: {
	nodeIntegration: true,
	enableRemoteModule: true, //don't miss the comma
	contextIsolation: false //this had to be added!
	},
	});
```

With this the installation was successful on Bullseye (image from 04-04-2022). 

Cheers Tim

rtulke, 2023-03-20:
This also works for me @reiseradeln thx!

Latest Raspbian OS, Release installed: 20.03.2023 

`root@raspberrypi:~# cat /etc/debian_version
11.6`

`root@raspberrypi:~# npm -v
7.5.2`

`pi@raspberrypi:~/TeleFrame $ npm version
{
  teleframe: '3.2.3',
  npm: '7.5.2',
  node: '12.22.12',
  v8: '7.8.279.23-node.57',
  uv: '1.40.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.17.1',
  modules: '72',
  nghttp2: '1.43.0',
  napi: '8',
  llhttp: '2.1.6',
  http_parser: '2.9.4',
  openssl: '1.1.1n',
  cldr: '37.0',
  icu: '67.1',
  tz: '2019c',
  unicode: '13.0'
}`

matzetronic, 2023-08-02:
> Hi @JanSHUP,
>
> I just tried to install TeleFrame on a new Raspberry Pi. In the end it worked.
>
>     1. Update the Pi
>        `sudo apt-get update`
>        `sudo apt-get upgrade`
>
>     2. Install Node [...]

For me these installation instructions failed because electron 25.x did not start because of strange errors like
```
../../buildtools/third_party/libc++/trunk/include/vector:1471: assertion __n < size() failed: vector[] index out of bounds[3650:0802/121144.157697:ERROR:gpu_process_host.cc(954)] GPU process exited unexpectedly: exit_code=6
```

So I removed the globally installed electron (v25):
`sudo npm remove electron -g`

And also globally installed v8 (if you have it installed):
`sudo npm remove v8 -g`

And **instead of instruction 4. from above** do the following:
4. Install lts version of node (I think this is missing in the list above):
`sudo n lts`

Then in your teleframe folder you can try to start it via:
`DISPLAY=:0 npm start`
This will use electron 13.x in your TeleFrame/node_modules folder. As this will be already installed by running the installer or `npm install` inside your TeleFrame folder. Also I don't understand the recommendation to have it installed globally. Then you have it installed twice, once globally (in /usr/local/lib/node_modules and in TeleFrame/node_modules) which is quite confusing.

```
npm version
{
  teleframe: '3.2.3',
  npm: '9.6.7',
  node: '18.17.0',
  acorn: '8.8.2',
  ada: '2.5.0',
  ares: '1.19.1',
  brotli: '1.0.9',
  cldr: '43.0',
  icu: '73.1',
  llhttp: '6.0.11',
  modules: '108',
  napi: '9',
  nghttp2: '1.52.0',
  nghttp3: '0.7.0',
  ngtcp2: '0.8.1',
  openssl: '3.0.9+quic',
  simdutf: '3.2.12',
  tz: '2023c',
  undici: '5.22.1',
  unicode: '15.0',
  uv: '1.44.2',
  uvwasi: '0.0.18',
  v8: '10.2.154.26-node.26',
  zlib: '1.2.13.1-motley'
}
pi@teleframe:~/TeleFrame $ npx electron -v
v13.6.6
```

For me it works with these versions on a pi3 using bullseye, hope it helps 👍🏻