LukeSkywalker92/TeleFrame #188: Black screen with logo on Debian

celphicious, 2022-05-25:
Hi Lukas,

when I start up TeleFrame I only see the black screen with white TeleFrame logo.
My bot is connected, since I can see the response in Telegram window and I can see my sent images
being saved into TeleFrame images folder locally. Altough pictures come in, the screen stays black with white "TeleFrame", won't react to "Alt" key or mouse or keyboard input.

My config file worked before in a raspberry based TeleFrame.
What bugs me, is the log output, it never says "Renderer started", but stays like this:

lars@bananapim2ultra:~/TeleFrame$ DISPLAY=:0 npm start

> teleframe@3.2.3 start /home/lars/TeleFrame
> electron .

2022-05-25T10:28:36.994Z [Main] info: Configuring for: Standard HDMI screen
2022-05-25T10:28:37.008Z [Main] info: Main app started ...
2022-05-25T10:28:38.284Z [Main] info: [AddonInterface] Load addons...
2022-05-25T10:28:38.293Z [Main] info: [AddonInterface] Initialize listeners...
2022-05-25T10:28:38.304Z [Main] info: [AddonInterface] Addons loaded
2022-05-25T10:28:38.382Z [Main] info: Add Admin-Actions
2022-05-25T10:28:38.386Z [Main] info: Bot created!
2022-05-25T10:28:38.479Z [Main] info: Bot started!
2022-05-25T10:28:38.945Z [Main] info: Using bot with name OmasFotoFrame_bot.

no logging info when a picture arrives, but still it is written into images folder.

I tried different Debian Distros (Raspbian, Debian, Armbian) on a Banana Pi M2 Berry.
I tried Nodejs versions 8,10,12,14 and 16. I updated npm to the latest version and tried without this also.
I tried automatic script install and also manual. On some distros I manually had to chown root:root and chmod 4755 the chromium-sandbox binary

I am working on this for hours now. What is a working combination of software here?
What am I doing wrong? Seems like something electron-related to me.

celphicious, 2022-05-25:
I just found this error in the Developer Tools Console of the chromium window TeleFrame runs in:
Uncaught ReferenceError: require is not defined at renderer.js:2

still trying to find out what it means.

celphicious, 2022-05-25:
okay, solution seems to be to set contextIsolation to false in main.js (after line 42) of TeleFrame,
this is a recent change in electron according to this: https://github.com/electron/electron/blob/main/docs/tutorial/context-isolation.md

// Create the browser window.
  win = new BrowserWindow({
    width: 1024,
    height: 600,
    webPreferences: {
      nodeIntegration: true,
      enableRemoteModule: true**,
      contextIsolation: false**
    },
  });

now it is running nicely! :)

Trenar, 2022-12-19:
Hi @MikkoTervala,
I have the same issue. However, I already have the version with the fix mentioned above... Any other ideas?

pm2 log is:

```
0|TeleFram | 2022-12-19T22:01:54.991Z [Main] info: Configuring for: Sunfounder HDMI screen
0|TeleFram | 2022-12-19T22:01:55.012Z [Main] info: Main app started ...
0|TeleFram | 2022-12-19T22:02:13.636Z [Main] info: [AddonInterface] Load addons...
0|TeleFram | 2022-12-19T22:02:13.638Z [Main] info: [AddonInterface] Initialize listeners...
0|TeleFram | 2022-12-19T22:02:13.640Z [Main] info: [AddonInterface] Addons loaded
0|TeleFram | 2022-12-19T22:02:13.741Z [Main] info: Add Admin-Actions
0|TeleFram | 2022-12-19T22:02:13.744Z [Main] info: Bot created!
0|TeleFram | 2022-12-19T22:02:13.746Z [Main] warn: Keyboard controls are disabled
0|TeleFram | 2022-12-19T22:02:15.272Z [Main] info: Scheduler started ...
0|TeleFram | 2022-12-19T22:02:15.276Z [Main] info: Bot started!
0|TeleFram | 2022-12-19T22:02:16.987Z [Main] info: 
0|TeleFram | 2022-12-19T22:02:20.330Z [Main] info: Using bot with name XXXXXXXXXXXXX.
```